don/cordova-plugin-ble-central #362: Fail connect ble return error peripheral disconnected

chubzchoc, 2017-02-20:
I do scan method and the other 2 times..
first, i scan bluetooth and then connect, write data and disconnect. then i scan new bluetooth and want to connect with it.  But when, i use ble.connect, i get error message : "peripheral disconnected". 
At the first section, connecting bluetooth success, but why at last section, i can't connect with ble device ?
return error happen when I call ble.connect. 
how to solve this? thank you

don, 2017-02-20:
Are you seeing on iOS or Android?

This happens every time with different devices, I'd look at something in your code.
If it only happens with certain devices, I'd look into those devices.

chubzchoc, 2017-02-21:
I'm using in Android..

chubzchoc, 2017-02-21:
Hi Don, 
I have solved this issue.. 
I've test my code to connect another BLE device and it's work..
When I get error, I using TI Sensortag.
thanks

don, 2017-02-21:
@chubzchoc glad you solved it

closed 2017-02-21 by don

nadeem2378, 2018-01-30:
@chubzchoc what is TI Sensortag? i am having the same error.

mburger81, 2018-02-14:
We got this error also many many many times.

If we try to connect to device we get this error and in this case we have to retry many times to connect to devices, after several tentatives finally it works.

We have this problem on many different devices, if we do a connection over android interface, at least there is never showing a connection error and it is very fast.

So what could the problems be for that?