don/cordova-plugin-ble-central #367: didUpdateValueForCharacteristic throwing NSInvalidArgumentException

soyelporras, 2017-03-06:
Hi don, first of all your plugin is wonderful. I have been using it for over a year now.

I have an iOS app that interacts with a BLE device that needs to do pairing before you can write or read from profiles.

On the very first time that I read a profile without the device being paired. The method didUpdateValueForCharacteristic  is throwing the following in the iphone's console log: *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[1]'

After spending a couple of days debugging with your plugin, I managed to solve it by adding the following inside didUpdateValueForCharacteristic:

if(readCallbackId) {
				NSData *data = characteristic.value; // send RAW data to Javascript
				CDVPluginResult *pluginResult = nil;
				
				if (error) {
					NSLog(@"%@", error);
					pluginResult = [CDVPluginResult resultWithStatus:CDVCommandStatus_ERROR messageAsString:[error localizedDescription]];
				} else {
					pluginResult = [CDVPluginResult resultWithStatus:CDVCommandStatus_OK messageAsArrayBuffer:data];
				}
				
				[self.commandDelegate sendPluginResult:pluginResult callbackId:readCallbackId];
				[readCallbacks removeObjectForKey:key];
			}

My question is why this wasn't implemented like you are doing inside didWriteValueForCharacteristic.

I hope this one helps anyone

don, 2017-03-07:
I'm glad you found a solution. Can you submit a pull request?

> My question is why this wasn't implemented like you are doing inside didWriteValueForCharacteristic.
I don't know, it was probably just an oversight.

soyelporras, 2017-04-06:
Hi don, thanks for getting back to me, sorry for the late response. I just submitted a pull request. Let me know if I missed something, this is actually my first time playing around with github

thanks

don, 2017-04-07:
Thanks!

See #385

closed 2017-04-07 by soyelporras