don/cordova-plugin-ble-central #368: Windows 10 support

VinceFa, 2017-03-07:
Hi ! I would like to know if Windows support (especially Windows 10) is still planned.

don, 2017-03-07:
@VinceFa currently Windows 10 doesn't support scanning for peripherals yet, you need to pair BLE peripherals before they can be used. 

Microsoft is working on this and scanning functionality is supposed to be in [Windows 10 preview SDK](https://www.microsoft.com/en-us/software-download/windowsinsiderpreviewSDK). 

So soon it will be possible to implement this for Windows 10. I'll probably add this if I need it for a project or someone hires me to add it for their project.

VinceFa, 2017-03-08:
Thanks for your answer ! Is the scanning functionality the only missing part of the plugin (for W10) ? Or do you mean that you will add all the W10 support later ? I have a doubt since there's no mention at all about Windows  in the readMe file. Anyway, thanks a lot for your plugin :)

don, 2017-03-16:
Currently any Windows code is stubs for PhoneGap Developer App, nothing actually works.

The Windows Bluetooth APIs currently require manual pairing of BLE devices from Windows, so as it stands the scanning feature of the plugin can not be implemented on Windows. Windows 10 preview SDK adds the new API calls needed to implement all the features.

don, 2018-06-08:
See #470

one0410, 2019-01-08:
Hello Don, any update for windows support ?

mexxwill, 2019-12-02:
@don any news update this? :)

peitschie, 2022-12-22:
The Windows platform has been deprecated as a cordova platform.
See https://github.com/apache/cordova/blob/master/deprecated.md#deprecated-platforms

closed 2022-12-22 by peitschie