don/cordova-plugin-ble-central #369: Advertising data on android

twinsit, 2017-03-07:
Hi,

I have again a question.
I am able to find devices, but I need to select only a certain type of devices, based on the Advertising PDU (because I only need non connectable devices).

But where can I find this. When I get the advertisement data, and I convert it, then I have
`{"0":2,"1":1,"2":6,"3":2,"4":10,"5":0,"6":7,"7":2,"8":3,"9":24,"10":2,"11":24,"12":4,"13":24,"14":5,"15":9,"16":73,"17":84,"18":65,"19":71,"20":0,"21":0,"22":0,"23":0,"24":0,"25":0,"26":0,"27":0,"28":0,"29":0,"30":0,"31":0,"32":0,"33":0,"34":0,"35":0,"36":0,"37":0,"38":0,"39":0,"40":0,"41":0,"42":0,"43":0,"44":0,"45":0,"46":0,"47":0,"48":0,"49":0,"50":0,"51":0,"52":0,"53":0,"54":0,"55":0,"56":0,"57":0,"58":0,"59":0,"60":0,"61":0} `.

Is there a certain order in this, so that I can know what this information means in fact? On iOs I can find it, but not on Android

Thanks a lot!

Wouter

don, 2017-03-07:
@twinsit you need to decode this info on Android using the [GAP type constants](https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile). It would be good to have some helper functions that did this on Android, but I haven't added them yet. See #192.

Here's some code that might help you. I [parse the advertising info into a map](https://github.com/don/ITP-BluetoothLE/blob/887511c375b1ab2fbef3afe210d6a6b7db44cee9/phonegap/thermometer_v2/www/js/index.js#L18-L39) and then get the service data to retrieve a [characteristic value that is being broadcast](https://github.com/don/ITP-BluetoothLE/blob/887511c375b1ab2fbef3afe210d6a6b7db44cee9/phonegap/thermometer_v2/www/js/index.js#L93-L103).

twinsit, 2017-03-08:
Ok, so if I am correct I need to convert it, and then find out the data key value that I need. So I only need to find out the correct key...

don, 2017-03-16:
Yes, you need to decode the bytes into something usable. It's tedious but not impossible. The specs and sample code should help. Also see how [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) interprets the advertising data.

closed 2017-03-16 by don