don/cordova-plugin-ble-central #372: Cannot connect to device after pairing

peteb5893, 2017-03-10:
I am working on an application for Android that will allow me to connect to a bluetooth enabled electricity meter, and I am able to detect it using ble.scan, then using ble.connect this displays a pairing key to connect the phone to the meter. Entering the key does pair the devices, and both my phone settings and the meter show the devices are paired, however, the connect method seems to break, showing neither success or failure output to the console. Has anyone else on here had similar issues and know of a solution?
Any help would be greatly appreciated.

Also I should mention the code I have written works fine for connecting to a Fitbit Charge HR, and I have been able to read off the Device Information Manufacturer's name charateristic "Fitbit" and display the value to the console. The fitbit however does not require a pairing key.

After scanning and generating a list of the available devices, click a connect button next to the device you wish to connect to. This works as expected for the Charge HR, but the console only displays the first output "Trying to connect..." when connecting to the electricity meter that requires the pair key.
```
$scope.connect = function (device_id) {
      console.log("Trying to connect to " + device_id);
      ble.connect(
        device_id,
        function (res) {
          var dev = res.name || res.id;
          var str = JSON.stringify(res, null, 4);
          console.log(str);
          console.log("Connected to " + dev);
        },
        function (err) {
          console.log("Error: " + err.errorMessage);
        }
      );
    }
```
Now I want to view a device specific page, so I click a button next to the device name that calls the following.
   ```
$scope.viewDetails = function (device_id) {
      ble.isConnected(
        device_id,
        function () {
          console.log("You are connected to " + device_id);
          $state.go('device', { 'id': device_id });
        },
        function () {
          console.log("Error: Not connected to any device");
        }
      );
    }
```
Finally I pass in the parameters for the manufacturer name characteristic, and the Charge HR device returns "Fitbit" as expected.
```
$scope.getManufacturer = function (device_id) {
      var serv_uuid = '180A'; // device information service
      var char_uuid = '2A29'; // manufacturer name characteristic
      ble.read(device_id, serv_uuid, char_uuid,
        function (data) {
          console.log(bytesToString(data));
        },
        function (err) {
          console.log("Error: " + err);
        }
      );
    }
```

Thanks in advance for any help or advice.

Peter

don, 2017-03-16:
What does `adb logcat` say while it's attempting to connect?

Can you connect to the meter using a tool like [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en)?

peteb5893, 2017-03-20:
We ended up using your other plugin, Bluetooth Serial to connect. This issue can be closed

closed 2017-03-20 by peteb5893

manish-katti, 2019-04-05:
Hi, 
I am using "cordova-plugin-ble-central": "^1.2.2" , i am facing this issue.
The device fails to connect if it is already paired, if I try to connect otherwise it works.