don/cordova-plugin-ble-central #374: Example of using promises in ble-central

LarsBuur, 2017-03-14:
I am looking for a simple example of using the ble-central with promises. Does anyone in here have an example they would like to share?

don, 2017-03-15:
@LarsBuur I don't have any public examples

There are wrapper functions that return promises in the javascript. See [ble.withPromises](https://github.com/don/cordova-plugin-ble-central/blob/master/www/ble.js#L166)

You might find some examples of people using the [Ionic2 wrappers](https://ionicframework.com/docs/v2/native/ble/) too.

loopezz, 2017-03-15:
With Ionic 2, can the startNotification observable be converted to a promise? I have to do a write and wait for a response on a notification observable.

don, 2017-03-15:
No, notification callbacks are many times. (Every time there is a notification.) Promises can only be resolved one time.

loopezz, 2017-03-16:
Hmm... Do you have an idea how to wait for the notify observable to be called after a write? At the moment I set a variable to null before a write and wait for this variable to be set by polling a function.

`
let readCharacter: string = null;

// start notification after connect
startReadingCharacter(){
	let that = this;

	BLE.startNotification(this.connectedDevice.id, this.serviceUUID, this.txUUID).subscribe(character => {
		that.readCharacter = that.bytesToHex(character);
	});
}

// send file packets (packets is an array filled with ArrayBuffers)
sendFilePackets(packets: any, index: number){
	return new Promise((resolve, reject) => {
		if(this.readCharacter){
			// check readCharacter and do some custom actions
			// example: check if ACK, increment index by 1, etc. Check if there are more packets to be sent. Reject by an error, resolve if completed, resolve* for trying again and resolve* for writing next data
			// * = resolve by calling this function
		} else {
			// should reject if there will be no response after x seconds
			setTimeout(() => {
				resolve(this.sendFilePackets(packets, index));
			}, 1000);
		}
	});
}
`

You will get the idea, I guess.

don, 2017-03-16:
I think you need to change how you're doing this. Disconnect the writing of data and notifications. A notification is an event, when you get it do something. (Maybe that something is send the next packet.)

If you can use write instead of writeWithoutResponse, use the success callback from write to know that you've written the data. If you have to use writeWIthoutResponse, assume the write works. Pause at least 30ms between writes. It looks like you're trying to send some sort of serial data over BLE, which can work but is slow and clunky.

closed 2017-03-16 by don