don/cordova-plugin-ble-central #375: Scan during connection?

lcrocker, 2017-03-15:
I have a device that does connectable advertisements until it is connected, then it advertises in non-connectable mode during the connection, then back to connectable after disconnecting.

scan() sees these devices just fine, and gets data from them. But after it connects to one, scan() seems to get advertisements from others, but not the one it's connected to.  I can see on a sniffer that it's still advertising, but scan() seems to be ignoring it.

cordova 6.5.0
ble-central plugin 1.1.4
android platform 5.2.2
android 6.0 (Moto G)

lcrocker, 2017-03-15:
Ah...just noticed that scanWithOptions(...{ reportDuplicates: true }...) doesn't have this issue. I guess you can call that an API choice, then.

don, 2017-03-16:
@lcrocker scanWithOptions is the solution for this

closed 2017-03-16 by don