don/cordova-plugin-ble-central #376: Connect error

loopezz, 2017-03-17:
At the moment I can't connect to my Bluetooth module with a Galaxy S6 Edge device (SM-G925F; Android 6.0.1). However, the same code works on a Galaxy XCover 3 (SM-G389F; Android 6.0.1).

This is some debug information on Galaxy S6 Edge.
Android Studio:
```
03-17 16:46:33.891 25393-25887/com.xxxxx D/BLEPlugin: action = connect
03-17 16:46:33.891 25393-25887/com.xxxxx D/BluetoothGatt: connect() - device: D8:80:39:F0:ED:5F, auto: false
03-17 16:46:33.891 25393-25887/com.xxxxx D/BluetoothGatt: registerApp()
03-17 16:46:33.891 25393-25887/com.xxxxx D/BluetoothGatt: registerApp() - UUID=9716a3a1-a470-4286-86ca-bb61d66fbf96
03-17 16:46:33.941 25393-25608/com.xxxxx D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
03-17 16:46:33.941 25393-25887/com.xxxxx W/PluginManager: THREAD WARNING: exec() call to BLE.connect blocked the main thread for 55ms. Plugin should use CordovaInterface.getThreadPool().
03-17 16:46:34.231 25393-25638/com.xxxxx D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=8 device=D8:80:39:F0:ED:5F
03-17 16:46:34.271 25393-25638/com.xxxxx D/BluetoothGatt: cancelOpen() - device: D8:80:39:F0:ED:5F
03-17 16:46:34.291 25393-25887/com.xxxxx D/BLEPlugin: action = disconnect
03-17 16:46:34.291 25393-25887/com.xxxxx D/BluetoothGatt: cancelOpen() - device: D8:80:39:F0:ED:5F
03-17 16:46:34.301 25393-25638/com.xxxxx D/BluetoothGatt: close()
03-17 16:46:34.301 25393-25638/com.xxxxx D/BluetoothGatt: unregisterApp() - mClientIf=8
03-17 16:46:34.321 25393-25887/com.xxxxx E/PluginManager: Uncaught exception from plugin
                                                              java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.BluetoothGatt.close()' on a null object reference
                                                                  at com.megster.cordova.ble.central.Peripheral.disconnect(Peripheral.java:89)
                                                                  at com.megster.cordova.ble.central.BLECentralPlugin.disconnect(BLECentralPlugin.java:353)
                                                                  at com.megster.cordova.ble.central.BLECentralPlugin.execute(BLECentralPlugin.java:166)
                                                                  at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:117)
                                                                  at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:98)
                                                                  at org.apache.cordova.PluginManager.exec(PluginManager.java:132)
                                                                  at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:57)
                                                                  at org.apache.cordova.engine.SystemExposedJsApi.exec(SystemExposedJsApi.java:41)
                                                                  at org.chromium.base.SystemMessageHandler.nativeDoRunLoopOnce(Native Method)
                                                                  at org.chromium.base.SystemMessageHandler.handleMessage(SystemMessageHandler.java:41)
                                                                  at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                  at android.os.Looper.loop(Looper.java:158)
                                                                  at android.os.HandlerThread.run(HandlerThread.java:61)
03-17 16:46:34.321 25393-25393/com.xxxxx D/SystemWebChromeClient: file:///android_asset/www/build/main.js: Line 44325 : connect error
03-17 16:46:34.321 25393-25393/com.xxxxx I/chromium: [INFO:CONSOLE(44325)] "connect error", source: file:///android_asset/www/build/main.js (44325)
```
Chrome:
```
errorMessage: "Peripheral Disconnected"
id: "D8:80:39:F0:ED:5F"
name: "XXXXX"
```

Code
```
scanDevices(){
	let that = this;

	return new Promise((resolve, reject) => {
		that.discoveredDevices = [];
		that.disconnectDevice();

		BLE.startScanWithOptions([], { reportDuplicates: false }).subscribe((discoveredDevice: any) => {
			if(/Vacuum_/g.test(discoveredDevice.name)){
				that.discoveredDevices.push(discoveredDevice);
			}
		}, (error: any) => {
			console.log('scan error', error);
		});

		setTimeout(() => {
			BLE.stopScan();
			resolve(that.discoveredDevices);
		}, 5000);
	});
}
	
connectDevice(selectedDevice: any){
	let that = this;

	return new Promise((resolve, reject) => {
		BLE.connect(selectedDevice.id).subscribe((peripheralData: any) => {
			console.log('connect success', peripheralData);
			that.connectedDevice = peripheralData;
			//that.startReadingCharacter();
			resolve();
		}, (error: any) => {
			console.log('connect error', error);
			that.disconnectDevice();
			reject();
		});
	});
}
```

Do you know what could cause this problem? Thanks in advance!

don, 2017-03-20:
The stack trace looks like you're calling disconnect on something that's already disconnected.

loopezz, 2017-03-22:
It's not related to that issue to be honest. I tried to establish a connection multiple times and I only try to disconnect if a peripheral is known (set in the connect success callback). However, only the connect error callback is being called on the S6 Edge device.

Functions:
```
discoveredDevices: any = [];
connectedDevice: any = null;

scanDevices(){
	let that = this;

	return new Promise((resolve, reject) => {
		that.discoveredDevices = [];
		that.disconnectDevice();

		BLE.startScanWithOptions([], { reportDuplicates: false }).subscribe((discoveredDevice: any) => {
			that.discoveredDevices.push(discoveredDevice);
		}, (error: any) => {
			console.log('scan error', error);
		});

		setTimeout(() => {
			BLE.stopScan();
			resolve(that.discoveredDevices);
		}, 5000);
	});
}
	
connectDevice(selectedDevice: any){
	let that = this;

	return new Promise((resolve, reject) => {
		BLE.connect(selectedDevice.id).subscribe((peripheralData: any) => {
			console.log('connect success', peripheralData);
			that.connectedDevice = peripheralData;
			resolve();
		}, (error: any) => {
			console.log('connect error', error);
			that.disconnectDevice();
			reject();
		});
	});
}

disconnectDevice(){
	let that = this;

	if(that.connectedDevice){
		console.log('disconnect device');
		BLE.disconnect(that.connectedDevice.id);
		that.connectedDevice = null;
	}
}
```

The disconnect seems to be called somewhere else, because the if statement blocks my disconnect (no log message will be displayed). This happens once in several times.

The main issue remains: the S6 Edge device can't get a connection, while the other Galaxy XCover 3 device can.

don, 2017-03-24:
This is probably device specific. Can you use [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) on the S6 Edge and connect to your peripheral?

You should not call disconnectDevice() when BLE.connect is rejected. The promise is rejected because the device disconnected.

loopezz, 2017-04-05:
Hi Don,

Thanks for your answer.

It has been a while for now. I guess it's indeed device specific. Even with nRF Connect the S6 device won't connect to the Bluetooth module. It's still a strange problem... I'm afraid it requires more debugging.

You may close this issue, because it's not a bug in your plugin.

closed 2017-04-05 by don