don/cordova-plugin-ble-central #379: Bluetooth device becomes invisible after it has been paired...

fraxool, 2017-03-24:
Hello,

I think it's a duplicate of #372 which has been closed. But the issue is still here. 

If you use a Bluetooth device - like a smart watch - which can be paired with an iPhone, the Bluetooth device is not visible anymore. So it's not possible to connect, to write or to notify.

I'm using the last version (1.1.4) of the plugin.

Hope it will be solved soon!

Thanks,
Axel

don, 2017-03-24:
If you have the UUID of the device you should be able to connect.

[CBCentralManager:retrieve​Connected​Peripherals​With​Services:​](https://developer.apple.com/reference/corebluetooth/cbcentralmanager/1518924-retrieveconnectedperipheralswith?language=objc) should return connected devices *if* you specify the service.

Android has [BluetoothManager.getConnectedDevices](https://developer.android.com/reference/android/bluetooth/BluetoothManager.html#getConnectedDevices(int))

The fix is probably to add a flag to [ble.startScanWithOptions](https://github.com/don/cordova-plugin-ble-central#startscanwithoptions) to return connected devices. Or perhaps to add a getConnectedDevices API call. The new API call is tricky because iOS needs a service but Android doesn't.

fraxool, 2017-03-24:
Thanks for the reply.

`If you have the UUID of the device you should be able to connect.`

I have the UUID. So with this UUID, I should be able to connect without adding a new API call like "getConnectedDevices" ? I'm not really sure how to connect if I can't "scan" the device. 
Should I just use the function "BLE.connect()" with the known UUID if the device has previously been paired ? 

Because I have tested to check if the device is connected (with the isConnected function) and it returned false.

don, 2017-03-24:
For iOS the UUID is generated so there's not a good way to get this without the new API. You should be able to disconnect the other app and scan to get the UUID for testing.

Ultimately scan should probably return connected devices. It'd be great if you could take a stab at implementing this.

florianchappuis, 2017-06-09:
Hi, I have the same issue than you fraxool? Have you managed to find a solution?

I have a peripheral bluetooth low energy (Nordic NRF51), I'm pairing up with it with my smartphone from the settings interface. They are bonded together, if i call scan() method, it isn't found. If I call connect method with the right MAC address, it says Peripheral MAC_ADDRESS not found. If i try to read a service, it returns : 
[INFO:CONSOLE(1362)] "ERROR", source: file:///android_asset/www/build/main.js (1362)

Could you help me?

fraxool, 2017-06-09:
@florianchappuis I personally never found a solution so I ended up using another library (evothings) which works well with pairing.

JonasWestAlro, 2017-11-15:
Hi, 

Still no news on this one? :)

ddugue, 2017-11-15:
Hey guys,
Is this an issue on both iOS and Android?
By the way, the plugin requires the device to have been scanned before being connected to.

JonasWestAlro, 2017-11-15:
I haven't tried on Android yet. 
I am running towards a deadline, so I will try the evothings as well :)

florianchappuis, 2017-11-15:
I used the evothings library, which i had to adapt because it way in javascript and ionic is in typescript, but it worked pretty well. Maybe they updated it. It was for a school project so i don't use it anymore

JonasWestAlro, 2017-11-15:
Yeah, I also had be make some small adjustments to make it work. Though when I am scanning, connected devices doesn't popup :( 

Did you just connect directly or something? @florianchappuis

don, 2018-06-11:
On iOS, the plugin still needs to know about the UUID. Try [ble.peripheralsWithIdentifiers](https://github.com/don/cordova-plugin-ble-central#peripheralswithidentifiers) or [ble. connectedPeripheralsWithServices](https://github.com/don/cordova-plugin-ble-central#connectedperipheralswithservices) if you don't want to scan.

Note that on recent versions of the plugin with Android, you connect using a MAC address without scanning.  Added [ble.bondedDevices](https://github.com/don/cordova-plugin-ble-central#bondeddevices) for Android.

closed 2018-06-11 by don