don/cordova-plugin-ble-central #381: where are the pages ?

alexdoso, 2017-04-01:
Hi Don, I would also like to ask you how does your code knows about 'mainPage' and 'detailPage' since a variable calling those elements is not defined anywhere in your code (except the html one, but that doesn't count). Thanks!

`  showMainPage: function() {
        mainPage.hidden = false;
        detailPage.hidden = true;
    },
    showDetailPage: function() {
        mainPage.hidden = true;
        detailPage.hidden = false;
    },`

don, 2017-04-06:
[mainPage](https://github.com/don/cordova-plugin-ble-central/blob/master/examples/bluefruitle/www/index.html#L14) and detailPage are defined in the HTML. When DOM elements have an id, there's automatically a Javascript variable with the same name. Alternately you could use 

    var mainPage = document.querySelector('#mainPage');

or

    var mainPage = document.getElementById('mainPage');

closed 2017-04-06 by don

alexdoso, 2017-04-06:
where can I see that variable that automatically is being created with the same name of any div with an id? In which exact file ?