don/cordova-plugin-ble-central #382: Question about this plugin compatibility

zabojad, 2017-04-04:
I'm trying to use this plugin with a peripheral that has the following communication protocol specifications:

1. Communication: full duplex serial communication, 5V_TTL level, UART interface
2. Baud rate: 9600
3. Data bits: 8 bit
4. One start bit, one stop bit
5. To put the Bluetooth power on and reset, and initiatively sending a set of data to the controller every 106 ms, when the controller received this set of data, after verified with the test value successfully, it will send back the set of data.

The peripheral BT interface exposes only one service to write data and get notified. This service accept an input value of 13 bytes.

The problem I'm having is that the peripheral doesn't react to my inputs sent from the cordova ble central plugin. I've tried to get help from the peripheral manufacturer (from china so communicating with them in English is not easy at all). I've sent them a little test app (using this plugin) and here is what they've replied:

> After tested the latest version app, it can be sending data, and the controller can also receive the data, but there’s still with issue on the timing of sending data, the data must be sent in continuous，please strictly follow our protocol to operate, then the result should be fine.

I don't understand what they mean exactly... Do you see any reason in the pieces of information I've written above why the plugin would not work correctly with this peripheral ?

zabojad, 2017-04-06:
@don do you see anything that would be incompatible with this plugin here?

don, 2017-04-06:
@zabojad The communication specs you listed look like Bluetooth Classic rather than BLE.

Did the manufacturer get your test app to worK? I'm not sure what the manufacturer's reply means. Maybe you need to send your BLE data with precise timing since it's expecting it to look like Bluetooth Classic? If you can connect to the service and characteristic this plugin should work. Figuring out what data you need to send is the tricky part.

Do you have another app that works with the hardware? You could sniff traffic and see what it's sending.

zabojad, 2017-04-06:
@don I got more details and it seems this peripheral needs to be sent data every 106 ms (gosh, javascript setTimeout is not even that precise...). Here is what the chinese engineer replied:

> We tested the new apk for the bluetooth demo. it still not follow our rule, maybe there have some misunderstanding.
> For 106ms cycles time. that is for whole data.  whole data which mean 1 to 13 data as below
[ 0x8, 0x82, 0x0, 0x14, 0xFA, 0x64, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0E ]
the draft step like
> 1. whole data
> 2. stop 106ms
> 3. whole data
> 4. stop 106ms
> .
> .
> .
> .
> .
> .
> repeat and repeat, never stop.

I've tried that and it still do not work :(...

Do you think using `ble.writeWithoutResponse` instead of `ble.write` would make a difference here ?

> If you can connect to the service and characteristic this plugin should work. 

Well, I can scan and connect to the device, I get the advertising data correctly (does that prove it is BLE and not classic BT?) and I try to write to the right service as usual (the success callback of `ble.write` is always called)... I also start listening to notification on the right service but I do not receive anything while I should... It's like if the controller behind the bluetooth module would not work / respond.

> Figuring out what data you need to send is the tricky part.

I have a description of what data to send but it does not work. The chinese engineer also confirmed I send the right bytes...

And unfortunately, I do not have any sample application for that peripheral that I could sniff...

don, 2017-04-07:
I don't like that design where the precise timing is required. I guess you're stuck with the hardware? You might need to write some native code to send data on that precise schedule. I'd try writing a sample app with the sample data.

If the characteristic supports both writeWithoutResponse and write it probably doesn't matter. which you use. Perhaps writeWithoutResponse will help get the timing better from your side since the OS won't need to wait for a reply from the peripheral.

zabojad, 2017-04-19:
@don the peripheral I was using was damaged. They've sent me a new one and it's working perfectly now. Thank you for your answers anyway!

closed 2017-04-19 by zabojad