don/cordova-plugin-ble-central #383: No option to scan for services

igorjos, 2017-04-04:
Hi I went trough the documentation and all features that are implemented, but there is no option to scan for services and characteristics.

Also when I get a list of devices the advertising object is always empty. 

I'm using this on Android 5.1

don, 2017-04-06:
@igorjos [ble.scan](https://github.com/don/cordova-plugin-ble-central#scan) lets you filter scan results for devices advertising services.

After you [connect](https://github.com/don/cordova-plugin-ble-central#connect) to a peripheral, the plugin discovers all the services and characteristics on your peripheral and returns them to you in the connectSuccess callback. https://github.com/don/cordova-plugin-ble-central#peripheral-data

igorjos, 2017-04-07:
@don  Well I assume that it should work like that, but actually I always get {} empty object, while with other BT app like nRF connect I can get a list of all services and characteristics, while using the same device.

Firstly I though that it may be due Uint array or similar, then I try to run conversion from Uint to string, also I check the same thing without any conversion just displaying the raw data, but every time it is the same response : {} 

Also I believe that there should be a separate method that will do that separate call. 
e.g I like to list only the devices, later on when I connect to it etc, I want to run scan for available services etc.

don, 2017-04-07:
If you're getting an empty peripherals, it sounds like something is failing. Look in `adb logcat` and chrome://inspect to see if there are any errors.

v1r, 2017-04-07:
I have the same problem, I am using android kitkat 4.4.2 in a tablet and trying to detect an Iphone 5s device.. I got an empty list when calling in the success callback of 'startScan' method.. 
I am now thinking of using Xamarin to create my app as I cant find any bluetooth plugin that works.. 

Any help would be appreciated.

jimboidaho, 2017-04-14:
I am not sure if this will help you but I have a tablet that is version 4.4.2 and scan does not work for me as well.  I have an version 5.1 phone and scan works fine. I think there is a problem with the older version.

don, 2017-04-14:
@v1r @jimboidaho Scanning is finicky on old Android devices. Some can't filter scan results for 128bit uuids and others can't scan 16bit uuids. Try scanning for all uuids and manually filtering the results in JavaScript. This is a different issue than the services not showing up after connecting.

igorjos, 2017-04-15:
@don sorry for a bit delayed response, no errors are thrown while listing the devices available, but no characteristics or services are displayed. I have tested this on 5.1 Huawei P7, and on 6.0 Huawei P9.

hervethouzard, 2017-05-02:
Filtering by service during scan does not work on old Android versions...

closed 2017-08-22 by don