don/cordova-plugin-ble-central #384: After the connection. The device is off and the connection is on for about 30 seconds

ramirwm, 2017-04-06:
How can we do this only ** 5 or 3 seconds,** then the error collback function that the device is off and please make new connection.

Thanx

don, 2017-04-06:
@ramirwm I'm not sure I understand. Do you mean when a peripheral disconnects on Android it takes 30 seconds to call the failure callback from ble.connect?

marcelviabb, 2017-04-10:
yes we are connected with the BL Device. 

Now we take off the Device and can not check if the device connected. 

we must waite 20 seconds for the reply.

can we check if the BL Device connected after the Device is off

don, 2017-04-10:
This plugins relies on the Android Bluetooth implementation to notify you of disconnects. Some phones are really slow notifying about disconnects. 

Do you have a simple example of some code to reproduce the problem?

hdung912, 2017-05-13:
hi Don!
I have save issue when i connected my device (Miband 2) and get (characteristics) and i wait <30 sec device connected after this time, ==> i have message  error - Peripheral Disconnected.

this is my code -   
connectDevice(deviceID) {
    this.characteristics = [];
    this.ble.connect(deviceID).subscribe(data => {
      // console.log('peripheralData.characteristics ' + JSON.stringify(peripheralData.characteristics));
      var dev = data.name || data.id;
      let str = JSON.stringify(data, null, 4);
      console.log(str);
      console.log("Connected to " + dev);
      }, function (error) {
      console.log("disconnected Device " + JSON.stringify(error, null, 4));
      alert("disconnected Device " + JSON.stringify(error, null, 4));
    }
    );
  }

don, 2017-09-05:
Sometime Android takes a long time before the BluetoothGattCallback.onConnectionStateChange() is called for a disconnect.

closed 2017-09-05 by don