don/cordova-plugin-ble-central #386: BLE data

puneet9, 2017-04-07:
Hello Don,

I have connected to BLE devices using your code and my BLE device is connected with Accelerometer, I want to read accelerometer data according to the no. of HZ i am assigning to it i.e. if i set 20 Hz it should send 20 readings per second, But using array buffer it sometimes never broadcasts the desired amount of readings and it slowly decrease the density of readings with time and sometimes there are no readings recorded for that amount of time.What do you think the possible cause for this could be my devices broadcasting or something to do with array buffer.

Thanks
Puneet

don, 2017-04-10:
@puneet9 there's not enough info for me to tell what's wrong. I'd look at your peripheral code first, it sounds like the device isn't sending the data correctly.

puneet9, 2017-04-10:
Sure, following is the js file attached with this mail for connection with
ble, my ble should send data for accelerometer attached to it according to
the frequency i write on it, i.e if if write 15 hz on it , it should send
me 15 readings per second but sometimes it skips the whole second or sends
1 or 2 readings per second.

(function (global) {
  
    var p = 0;
    var q = 0;
    var X10, Y10, Z10, X11, Y11, Z11, X20, Y20, Z20, X21, Y21, Z21, message1, message2, Hertz, Timel,TimeR;
    var x1, y1, z1, x2, y2, z2, g, b, c, d, e, f, ax1, ay1, az1, ax2, ay2, az2;
    var data2 = 0xf;
    var data3 = 0;
    var data4 = 0;
    var countl = 0;
    var countr = 0;
    var arrayl = new Array();
    var arrayr = new Array();
    var left = new Array();
    var right = new Array();
    var xxr = new Array();
    var xxl = new Array();
    var leftxxl = new Array();
    var rightxxl = new Array();

    button3.onclick  = function () {  change(5)  };
    button4.onclick  = function () {  change(15) };
    button5.onclick  = function () {  change(10) };
    button6.onclick  = function () {  record(1)  };
    button7.onclick  = function () {  record(0)  };
    button8.onclick  = function () {  step(1)    };
    button9.onclick  = function () {  written()  };
    button10.onclick = function () {  counter()  };
    button11.onclick = function () {  leftc()    };
    button12.onclick = function () {  rightc()   };
    button13.onclick = function () {  show()     };
   
    var app = {

        scan: function () {
            alert("BLE Test1: scanning...");
            disconnect('C5:B6:C9:7D:93:82');
            disconnect('EE:F7:04:7D:A3:98');
            ble.scan([], 10, scanSuccess, onError);
            gif.style.visibility = "visible";
            
            p = 0;
            q = 0;
        }
       
    };
   
    
    function scanSuccess(device) {
        var deviceId = device.id;
        var devices = new Array();
       
        devices.push(deviceId);
        
       
        for (var i = 0; i <= devices.length; i++) {
            if (devices[i] == 'EE:F7:04:7D:A3:98' && 'C5:B6:C9:7D:93:82') {
               
                alert("Your Devices found");
                connectR('EE:F7:04:7D:A3:98');
                connectL('C5:B6:C9:7D:93:82');
               
            }
            else {
[... 710 lines cut ...]
                    fileEntry.createWriter(function (fileWriter) {
                        fileWriter.truncate(0);

                    });
                });
            });

            data3 = 1;
            alert("Recording Started");
        }
        else {
           
            data3 = 0;
            alert("Recording Stopped");
        }

    }
    function step(value) {
        text4.innerHTML = "Right:" + countr;
        text3.innerHTML = "Left:" + countl;
    }

    global.app = app;
    
})(window);

closed 2017-08-22 by don

chjchoi, 2018-01-18:
I used the CC2541 System and stack 1.4.2.2 for ble.
 It worked very good on Android 4.4.4 system.
 but on Android 5. it was very very slow...

I have found out and resolved the problem.
 CC2541 I used 16bit UUID such as 'fff0' 'fff1'
 I did not changed the uuid code for CC2541 firmware
 but I switched the service uuid and characteritcs uuid to 128bit uuid of Ionic apps.

then..slow problem was resolved.

Instead of 'fff0' 'fff1' I used on the Ionic apps
 0000FFF0-0000-1000-8000-00805F9B34FB
 0000FFF1-0000-1000-8000-00805F9B34FB

these uuid information were from other BLE scan app's displaying...
 I have not changed the 16bit uuid system on CC2541 frimware
 because when I use the 'fff0' 'fff1' it means the 128uuids above.

More exact result
 I should change the CC2541 uuid systems to 128bit
 and use TI BASE UUID
 F000FFF0-0451-4000-B000-000000000000
 F000FFF1-0451-4000-B000-000000000000
 and change the Ionic uuid also
 but I did not changed the CC2541 stack uuid.
 just use the 0000 ....34FB 128bit uuid only on the ionic app.