don/cordova-plugin-ble-central #388: Stuck connections when closing Android app.

mebrunet, 2017-04-10:
I'm seeing an issue on certain Android devices where closing the app doesn't terminate the BLE connections I had initiated.

**Steps to reproduce**: Open an app -> connect 2 or more peripherals -> close the app via `navigator.app.exitApp()` or by closing the process in the tab manager.

**Result**: One or more peripherals will remain connected. They are *stuck*... only toggling the Bluetooth will terminate the connection. 

**Expected result**: Closing the app should release all BLE resources.

On some devices, e.g. Nexus 5  - android 6.0.1, I see the expected result. On others, e.g. Axxon A1R - android 6.0.1, I don't. 

I'm digging into this in more details now, but I was wondering if anyone else has seen a similar issue.

don, 2017-04-10:
Are you seeing this in Cordova apps or in PhoneGap Developer App? If you're running in the PhoneGap developer app you need to manage connections since the app keeps running. In Cordova, all connections *should* be freed when the app closes.

mebrunet, 2017-04-10:
I remember our convo re: Developer App... but in this case the apps have been fully built-out using PGB.

don, 2017-04-10:
See if you can duplicate it. But if the connections are being held after the app is closed, that sounds like an OS issue.  Or perhaps the app doesn't really close? 

I'd enable the btsnoop_hci.log on a good device and a problematic device. Then use Wireshark to examine the logs and see if there's different behavior on disconnect for the working device vs the broken one. You might need to explicitly close connections on the phone before closing the app.

mebrunet, 2017-04-10:
Ya, I've pulled the `btsnoop_hci.log` and the `adb logcat` for the process. We're comparing now. 

I also find this very strange, and suspect the issue runs deep (like OS deep). We can easily recreate the bug with our app. But only on some phones... What's concerning, is that no amount of "force closing" the app (via process manager etc.) seems to release the connections. Only toggling the Bluetooth will solve the problem...

All this being said, if I use another BLE app, eg. Nordic's [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en), I can connect the same number of peripherals, close the app with the Android tab manager, and all BLE devices disconnect. (Even on the phones that wouldn't when we did the same thing with our app)

We're also seeing some Gatt/BLE null-pointer exceptions popping up in logcat, and I'm wondering if they're related. It might be that the library is causing errors that some manufacturers' flavour of Android / BLE drivers catch and clean up nicely, while others don't... leading to stuck connections.

don, 2017-04-10:
> We're also seeing some Gatt/BLE null-pointer exceptions popping up in logcat, and I'm wondering if they're related. It might be that the library is causing errors that some manufacturers' flavour of Android / BLE drivers catch and clean up nicely, while others don't... leading to stuck connections.

That is very possible. If you can track that down the details, it may be possible add some additional code to the plugin to work around it.

mebrunet, 2017-04-11:
Consistent way to recreate the bug:

## Setup
1) Create a new phonegap project
2) Modify `config.xml` to include this plugin from npm
3) Modify `index.html` to load script `js/bleTest.js` with contents:
```
var bleTest = {
    // replace MAC_X and NAME_X accordingly
    testDevices: [
        'MAC_1': 'NAME_1',
        ....
        'MAC_6': 'NAME_6', 
    },
    run: function () {
        ble.scan([], 10, (device) => {
            if (device.id in this.testDevices) {
                console.log('Attempting to connect: ' + this.testDevices[device.id]);
                ble.connect(device.id, console.log, console.error);
            }
        }, console.error);
    },
}
```
4) Build with PGB and install on an Android mobile device
5) Open the app with the mobile device in debug mode and USB connected to a computer, then use `chrome://inspect` to attach to the phonegap webview and get a remote console. 
6) Execute `bleTest.run()` in the console. (Make sure Bluetooth and Location services are on first)
7) After connections' success/error callbacks all fire, exit the app using Android tab/process manager.

## Result: 
*On some devices* connections get "stuck", and can only be terminated by toggling the Bluetooth on/off.

*On ALL devices* if the app is force closed *before* all the connection callbacks have fired... then connections will sometimes remain "stuck".

pooja-shah, 2017-05-09:
Have you find any solution for this issue? I am facing same issue with my cordova app.

mebrunet, 2017-05-18:
@pooja-shah - basically we found it to be an issue with the core Bluetooth API on the devices in question. i.e. we were able to recreate the issue with a native app using recommended connection/disconnection processes. We had to reorganize our app considerably as a result, limiting to one connection at a time on Android.  We also forked this repo and applied a few more fixes. See https://github.com/ddugue/cordova-plugin-ble-central

don, 2017-06-02:
@mebrunet thanks for the update. if you have generic fixes for the plugin send a pull request.

closed 2017-06-02 by don

nimeso, 2017-10-01:
Did this get merged? We are still having this issue.