don/cordova-plugin-ble-central #389: EXC_BAD_ACCESS

claudiovolpato, 2017-04-13:
hi Don,
when compiling your code with xcode 8.1 and run it on iOS 9.3.5 the following line of code:
  
 // check and handle disconnected peripherals
    for (CBPeripheral *peripheral in peripherals) { ....

to get : EXC_BAD_ACCESS

Can you help me? Thanks.

don, 2017-04-14:
I won't have access to iOS 9 devices until next week. Does this error happen every time?

claudiovolpato, 2017-04-14:
Hi Don,
yes every time :-(

claudiovolpato, 2017-04-14:
Seems to be an initialization problem for the NSMutableSet * peripherals variable.
  Unfortunately I do not know object-C :-(

claudiovolpato, 2017-04-26:
Ho Don, Did you find a minute for me? :-)

claudiovolpato, 2017-04-29:
Hi Don, i have found the problem:
in pluginInitialize i have substitute

    peripherals = [NSMutableSet set];
with
    peripherals = [NSMutableSet new];

now works.
Thanks anyway :-)

don, 2017-05-04:
@claudiovolpato thanks for updating me with the answer, I'll look at getting this into the next release

commit 432acd80 (2018-06-07): replace set with new Fixes #389
closed 2018-06-07 by don