don/cordova-plugin-ble-central #391: startScan doesn't return any BLE devices on Android 7

develxxx, 2017-04-19:
Hello, after updating my Nexus 5x to Android 7.1.2 the startScan function doesn't return BLE devices.
The BLE devices are visible in the Android Settings bluetooth section.

ADB LOGCAT output 

04-19 13:46:10.249 6831-6843/? D/BtGatt.GattService: start scan with filters
04-19 13:46:10.253 6831-6855/? D/BtGatt.ScanManager: handling starting scan
04-19 13:46:10.254 9870-10009/com.gwin.connect W/PluginManager: THREAD WARNING: exec() call to BLE.startScan blocked the main thread for 22ms. Plugin should use CordovaInterface.getThreadPool().
04-19 13:46:10.263 6831-6851/? D/BtGatt.GattService: onScanFilterEnableDisabled() - clientIf=5, status=0, action=1
04-19 13:46:10.263 6831-6851/? D/BtGatt.ScanManager: callback done for clientIf - 5 status - 0
04-19 13:46:10.264 6831-6855/? D/BtGatt.ScanManager: configureFilterParamter 500 10000 1 0

The THREAD WARNING message doesn't appear on my Android 5.1 phone and the BLE devices are enumerated. Thanks for help.

kirkryan, 2017-04-20:
HI, I have the same issue with my Moto G4 Plus. The plugin used to work on Android 6.0 but since updating to Android 7.0 the scan function doesn't return any devices. The BLE device is visible in the Android settings Bluetooth section.

shfitz, 2017-05-02:
Chiming in with what appears to be the same issue with a Nexus 6p. I don't have firsthand experience with an earlier version of the OS, so I can't comment on if it's unique to this phone or a recent update to Android 7.1.2.

hervethouzard, 2017-05-02:
Hello,

Is the location service activated?

shfitz, 2017-05-02:
Location services are activated. Devices are visible in the bluetooth settings and scanner apps, but not in the phonegap application.

develxxx, 2017-05-02:
Hello, 
**after enabling location service** on my Nexus 5x  (Android 7.1.2) by starting google maps the BLE devices were found by startScan ! It seems that the update affects the location service user settings. Thank you very much for this tip, hervethouzard !

closed 2017-05-02 by develxxx

arctouch-italowanderley, 2017-10-24:
I know that post is closed, but I've decided to post this here: 
```
Note: LE Beacons are often associated with location. In order to use BluetoothLeScanner without a filter, you must request the user's permission by declaring either the ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission in your app's manifest file. Without these permissions, scans won't return any results.
```
from: https://developer.android.com/guide/topics/connectivity/bluetooth-le.html