don/cordova-plugin-ble-central #394: Reconnecting BLE

AlexSm13, 2017-04-27:
We have a device that connects using BLE.

The first connection is successful.

But when the signal is lost, reconnection does not happen

Connected:

`ble.connect (deviceAddress, connectSuccess, connectFailed);`

If the connection is successful, I add an event to the button

```
function connectSuccess (characteristic) {
    log ("connectSuccess to" + characteristic.name);

    var fieldState = document.getElementById ("fieldState");
    fieldState.addEventListener ("click",
        function () {readData (uuids1.StateState, "field State", "textField"); });
}
```
When the connection fails, I interrogate the device every 10 seconds

```
function connectFailed (characteristic) {
    log ("connectFailed to" + characteristic.name);
    ble.disconnect (deviceAddress);
    reconnect = setTimeout (function tryToBack () {
        ble.connect (deviceAddress, sc, connectFailed);
        log ("Try Connecting !!");
        reconnect = setTimeout (tryToBack, 10000);
        },
    100);
}

function sc (data) {
    clearTimeout (reconnect);
    log ("Success reconnect to" + data.name);
    log ("with address" + deviceAddress);
    connectSuccess (data)
}
```
But when the device is in sight, it is connected but does not respond to a button click.

harishkumarr95, 2017-04-28:
Hello @AlexSm13,
I am also facing the same issue here. The connection stays for about an approximate of 30 - 45 mins and then gets disconnected and does not reconnect at all. The reconnection happens when I reset the Bluetooth device and turn off and turn on the native Bluetooth in the android device. It would be great if there is something that would make the reconnection automated as soon as the disconnection happens.

Golotus, 2017-06-20:
Why I can't code like this:

`ble.connect (deviceAddress, sc, connectFailed);
`
if I code this, I will get a error like this:

`Supplied parameters do not match any signature of call target.
`

Domvel, 2018-06-05:
There is a new method `autoConnect`. Is this your solution? Otherwise try a scan before connect. In my experience, BLE on Android is not mature. I also have some problems with connect and disconnect. 
btw. Don't forget the notifications on reconnect. Also the method 'autoConnect' will not automatically re-subscribe the notification. You have to this.

closed 2018-06-07 by don