don/cordova-plugin-ble-central #396: scan or startScan doesn't return anything on Android 7.1.2

shfitz, 2017-05-03:
This may or may not be related to issue #391 

Hardware : Nexus 6p running 7.1.2 
Issue : Calling scan and startScan do not return any devices.

I have location services activated and ble devices are visible in the bluetooth settings and scanner apps, but not in phonegap applications. 

The same code works on ios. Any thoughts.

don, 2017-05-03:
@shfitz Do you see any errors if you run "adb logcat"?

Can you send me the code so I can use it for debugging?

/cc @tigoe

tigoe, 2017-05-03:
@don https://github.com/tigoe/MakingThingsTalk2/tree/master/3rd_edition/chapter10/BleDatalogger is the project @shfitz is running.  I didn't see errors running Android 6, but he did in 7.

shfitz, 2017-05-04:
@don @tigoe  I didn't see any errors, but to be honest, I'm not sure where to look for them.

Log from starting up attached. It seems that the scan starts before i actively initiate it, which isn't what the code is supposed to do. 

[pgap_androidoutput.txt](https://github.com/don/cordova-plugin-ble-central/files/975469/pgap_androidoutput.txt)

don, 2017-05-04:
Can you see if you get different results using the 128 bit version of the UUID?

Replace FE10 with 0000FF10-0000-1000-8000-00805F9B34FB

https://github.com/tigoe/MakingThingsTalk2/blob/master/3rd_edition/chapter10/BleDatalogger/www/sketch.js#L17

shfitz, 2017-05-04:
I can check that in the morning, but when I don’t specify a device, nothing
populates the dropdown either. For example
ble.scan([], 5, discoverDevice, bleError);
returns nothing either.

https://github.com/tigoe/MakingThingsTalk2/blob/master/3rd_edition/chapter10/BleDatalogger/www/sketch.js#L17

don, 2017-05-04:
If "ble.scan([], 5, discoverDevice, bleError);" doesn't work the 128-bit UUID probably won't either. I'll try and duplicate the problem here. (It might take a few days.)

tigoe, 2017-05-05:
The drop-down doesn't populate until scan runs.

I don't have an Android 7 device, so nothing I can test with yet.

nicearma, 2017-05-08:
I have the same problem, i try to use scan or startScan and nothing happen, if i use enable and showBluetoothSettings functions seems to work fine.
I'm using ionic 2 with this plugin and my device is One Plus 3 with android 7.1.1

nicearma, 2017-05-10:
I'm trying to debug the app for fix the plugin, but the code source from the sdk and my phone aren't the same, but i can tell you that the findLowEnergyDevices is been call and the problem seem to be the startLeScan function call, i will try to find the source sdk code (ligne 2317 of BluetoothAdapter) that match with my device

edupaz2, 2017-06-15:
Hello.

I'm having the same problem.

Device: Mediatek; Android Android 7.0; Ionic 3;

Code in app:
`scan() {
    this.status = "Scanning";

    this.ble.scan([], 5).subscribe(
      (device: any) => { console.log("Device "); },
      (err) => { console.log("Error"); },
      () => { console.log("Finished"); }
    );
  }`

Attaching logcat output: 
[ble.logcat.txt](https://github.com/don/cordova-plugin-ble-central/files/1076865/ble.logcat.txt)

I saw in the logs MAC addresses being scanned, but I also saw this:
"PluginManager: THREAD WARNING: exec() call to BLE.scan blocked the main thread for 75ms. Plugin should use CordovaInterface.getThreadPool()."

edupaz2, 2017-06-16:
Hi again.

If you turn on the Location on your device, onLeScan callback starts to receive calls.

Regards.

don, 2017-08-22:
@shfitz @tigoe Sorry this took so long to track down. It looks like bad meta data in config.xml is causing this problem. I created a pull request with the updated config.xml to fix the issue https://github.com/tigoe/MakingThingsTalk2/pull/4

Below are the steps I used to reproduce the issue. If I repeated with [my fork](https://github.com/don/MakingThingsTalk2/tree/ble-central-issue-396) scan discovers devices.

Get the code
    mkdir ble-issue-396
    cd ble-issue-396
    git clone https://github.com/tigoe/MakingThingsTalk2.git

Start the data server

	cd ble-issue-396/MakingThingsTalk2/3rd_edition/chapter10/dataServer
	npm install
	npm start

Create the peripheral.

	Open a new terminal
    cd MakingThingsTalk2/3rd_edition/chapter10/Arduino101GSRSensor

Open Arduino101GSRSensor.ino with Arduino IDE 1.8.3 
Use the Board Manager to update/install Intel Curie Boards v2.0.2
Compile and deploy to Arduino 101

Compile the Cordova app

	cd MakingThingsTalk2/3rd_edition/chapter10/BleDatalogger
	
Change the http://192.168.0.10:8080 in index.html and sketch.js to match ip address of laptop
	
	cordova platform add android

Plug in Android phone & deploy the app

	cordova run android --device
	
App will prompt for access to location. Choose Allow.
Press the "Scan for devices"
When scan is complete, select the Arduino 101 from the list.

Versions
 * Node v 6.11.2
 * Cordova v 7.0.1
 * OS X 10.12.6
 * Android 7.1.2 running on Nexus 5X

closed 2017-08-22 by don