don/cordova-plugin-ble-central #398: Android uses cached service uuid.  Will not update.

T-Sneddon, 2017-05-04:
Android 5.0.2 caches custom service uuid and will not update when custom service is changed in Bluetooth device.  Various newer versions of Android do update.

I found this issue addressed on Stack Overflow at [http://stackoverflow.com/questions/22596951/how-to-programmatically-force-bluetooth-low-energy-service-discovery-on-android](http://stackoverflow.com/questions/22596951/how-to-programmatically-force-bluetooth-low-energy-service-discovery-on-android)

I'm not a Java programmer by any stretch of the imagination, but I was able to find a place to insert and test this solution in Peripheral.java and it worked.  I can now change the service uuid of the Bluetooth device and the plugin is able to see the new uuid.

Attached is a diff of what I did to Peripheral.java, and I just want to thank you for all you've done to make such a great ble plugin.  I've tried them all and this one is the best, and has far and away the best documentation.  Having good documentation and plenty of working examples is priceless when you're first getting started.

[Peripheral.diff.txt](https://github.com/don/cordova-plugin-ble-central/files/976545/Peripheral.diff.txt)

hervethouzard, 2017-09-26:
There's no working solution to clean this cache unfortunately.

don, 2018-06-07:
See #587

closed 2018-06-07 by don