don/cordova-plugin-ble-central #401: How to read BLE device data?

SaadeMotion, 2017-05-07:
I'm working with Estimote beacons, my beacons are the new versions with sensors. I'm trying to read the motion sensor, but i don't know how.

Here's the data:

```
{
  "id": "E9:BE:31:9E:59:E1",
  "advertising": {},
  "rssi": -43,
  "services": [
    "1800",
    "1801",
    "6c372000-d2cc-11e4-9a1f-0002a5d5c51b",
    "6c371000-d2cc-11e4-9a1f-0002a5d5c51b"
  ],
  "characteristics": [
    {
      "service": "1800",
      "characteristic": "2a00",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a01",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a04",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1801",
      "characteristic": "2a05",
      "properties": [
        "Indicate"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        }
      ]
    },
    {
      "service": "6c372000-d2cc-11e4-9a1f-0002a5d5c51b",
      "characteristic": "6c372001-d2cc-11e4-9a1f-0002a5d5c51b",
      "properties": [
        "WriteWithoutResponse"
      ]
    },
    {
      "service": "6c372000-d2cc-11e4-9a1f-0002a5d5c51b",
      "characteristic": "6c372002-d2cc-11e4-9a1f-0002a5d5c51b",
      "properties": [
        "Write",
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        }
      ]
    },
    {
      "service": "6c372000-d2cc-11e4-9a1f-0002a5d5c51b",
      "characteristic": "6c372004-d2cc-11e4-9a1f-0002a5d5c51b",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "6c371000-d2cc-11e4-9a1f-0002a5d5c51b",
      "characteristic": "6c371002-d2cc-11e4-9a1f-0002a5d5c51b",
      "properties": [
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        }
      ]
    },
    {
      "service": "6c371000-d2cc-11e4-9a1f-0002a5d5c51b",
      "characteristic": "6c371001-d2cc-11e4-9a1f-0002a5d5c51b",
      "properties": [
        "WriteWithoutResponse",
        "Write"
      ]
    }
  ]
}
```

i'm new to this bluetooth stuff, can u give me some idea how i can read this sensors?

genedragotta, 2017-05-29:
the service UUID should be before the characteristic UUID

genedragotta, 2017-06-01:
here's an example on reading the GATT battery service.....

try {
	ble.read(macAddress, "180F", "2A19", 
		function( arraybuf ) { 
			try {
				var strData = String.fromCharCode.apply(null, new Uint8Array(arraybuf));
				var bl = strData.charCodeAt(0);
				alert('battery level: ' + bl);
			} catch(err) { 
                                  alert('onConnect BLE.READ error: ' + err.message);  
                       }					
		}, onReadError
	);

} catch(err) { 
	alert('battery read err: ' + err.message); 
}

don, 2017-06-02:
It looks like Estimote changes the UUIDs that are broadcast to send motion data rather than using a Service and Characteristic to send motion data.

https://community.estimote.com/hc/en-us/articles/204971127-How-to-use-a-beacon-s-accelerometer-What-is-Motion-UUID-

Estimote recommends using Telemetry to get the motion data http://developer.estimote.com/sensors/estimote-telemetry/, but this also looks like broadcast data. 

There might be an Accelerometer service with a characteristic you can subscribe to for notifications. Devices like the TI SensorTag do this. I don't have an Estimote with me today, so I can't check. 

[Sensor Tag Accelerometer Documentation](http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User%27s_Guide#Movement_Sensor)
[Cordova Sensor Tag Example](https://github.com/don/cordova-plugin-ble-central/blob/master/examples/sensortag_cc2650/www/js/index.js#L85-L96)

closed 2017-07-07 by don

jayeshkorat18, 2018-04-02:
After successfully connect I try to read battery status.
Notable read battery status, My device characteristic for reading is : CCDLJC$,