don/cordova-plugin-ble-central #408: Scan Issue, Sometimes scan fail

hwanil, 2017-05-29:
Hi, I'm hwanil
First of all, I'm not good at English...

I'm so thank you don and your plugin.
This Plug-in very powerful and useful.

So, My problem is scan.

When I scan first, application scan ble device very well.

but when i re-try several time, application does not scan ble.

My scan code like this

`document.addEventListener("deviceready", function () {
            ble.startScanWithOptions([BLE_ID], { reportDuplicates: true }, function (peripheral) {
                ble.stopScan(
                    function() {
                        do Something
                    function() { console.log("stopScan failed"); }
                );
            }, function (reason) {
                alert("ERROR: " + reason);
            });
        }, false);`

Another BLE APP can scan device. 

I want to know what is problem.

I think app shut down suddenly, application cash will remain and this is can be problem factor.

sebj54, 2017-06-02:
I have a similar problem, I assume it's related. The difference between you and me is that I can't make it work when I explicitly set a service UUID like this:
```js
ble.startScan([serviceUUID], function(device) { console.log(device) })
```

If I remove the service UUID, it works correctly. Can you try to scan for your device without setting a service UUID?

don, 2017-06-02:
@hwanil Some Android devices aren't very good at scanning for service UUIDs. Try scanning for all devices.

If you've connected to a device, it typically stops advertising. There are cases where iOS or Android is connected to the device but your app can't see it. This might happen in a crash.

hervethouzard, 2017-06-21:
Filtering on services does not run on all Android versions (specially on old ones)

closed 2017-07-07 by don