don/cordova-plugin-ble-central #411: subscribing to iBeacon notfications without connection

hughbris, 2017-06-02:
TLDR; _are there successful iBeacon implementations which subscribe to a service?_

I've been told I don't need to connect iBeacon peripherals, and should be able to just listen for services without that step. I'm getting errors, which I will describe below.

So when I scan, I see the peripheral, though the advertising data property shows `{}`. That's background.

_(I scan with services=[] and use JS later to filter, I've also tried startScanWithOptions just for giggles)_

When I invoke `ble.startNotification()` and log the error, I see "Peripheral <mac> is not connected". For science, I've also tried `ble.read()` and it presents the same error.

Just to humour the library, I tried connecting to it using `ble.connect()` - the failure callback was invoked eventually with err == "Peripheral disconnected". I've tried connecting with another (non-iBeacon) peripheral with the same code and it succeeds.

My question is, do you know of iBeacon implementations where the code successfully follows these steps?:

* scan for iBeacon device
* start notifications for a service UUID

What should the process be for subscribing to iBeacons if I have these steps wrong?

I have tried the same process with almost identical results using [randdusing's library](https://github.com/randdusing/cordova-plugin-bluetoothle). I have not tried [petermetz's library](https://github.com/petermetz/cordova-plugin-ibeacon) mainly because it's hard to understand his docs (thank you @don for the absolute best docs BTW!), it's over a year since it was updated, and it has 108 outstanding issues!

I'd love to use this library, it's elegant and intuitive, but I can't easily tell if it's been tested/developed for beacons. Also I know very little about hardware. I'm mainly focused on (technical) interfaces and this process was provided by our hardware guy. I can see the target service ID in nRF Connect (as far as I can understand that app).

I'm only holding back on posting my code because it's super messy with all this rewriting and experimentation, and I'm hoping my words do a better job this time.

halindrome, 2017-06-02:
In general an iBeacon is only an advertiser.  In fact, the Apple iBeacon spec does not permit the iBeacon to do anything else!  And no - you cannot enable notifications from a service without first connecting.  At least, not as I read the Bluetooth SIG specifications.

The normal sequence when you want to subscribe to notifications from a device is:

Scan for a device signature
If found, connect to that device
On connection success, set up your notification listener and then write to the device to tell it that you want to activate notifications for whatever service.

Also, remember to have a disconnect handler sitting around to deal with cleaning up when the device goes out of range.

don, 2017-06-02:
> So when I scan, I see the peripheral, though the advertising data property shows {}
If you console log an ArrayBuffer it will appear empty even though there's data. Try converting to Uint8Array then logging.

This plugin isn't intended for scanning beacons. You can get raw iBeacon data on Android, but you'll need to parse the data yourself. On iOS, they (annoyingly) remove the iBeacon data from the advertising info. On iOS you need to use CoreLocation to get this.

startNotification didn't work for you since you need to be connected and you can only subscribe to characteristics.

Generally I recommend people use https://github.com/petermetz/cordova-plugin-ibeacon for Beacons. Most of the beacon stuff I've done personally is native code, but we do cover Peter's plugin in Chapter 10 of [Make:Bluetooth](https://www.amazon.com/Make-Bluetooth-Projects-Raspberry-Smartphones/dp/1457187094).

hughbris, 2017-06-02:
@halindrome: thanks, the plugin (or lower down) refuses connection though (if I wasn't clear).

@don Thanks, it's a great prompt response (ahem _cough cough_ [evothings](https://github.com/evothings/cordova-eddystone/issues/18)), unfortunately not quite the _answer_ I hoped for :/

> Try converting to Uint8Array then logging.

I actually suspected from reading other issue responses it was simply an output thing with the advertising, so should have omitted that detail. Only muddied things.

> This plugin isn't intended for scanning beacons.

Perhaps your excellent README could make that slightly clearer to ward off desperate fools like me. Happy to provide a small patch/PR if you agree.

I'll take a look at the Make article and see if that makes tinkering with Peter's library any more worthwhile. No other options I can see.

Will close this. Thanks again!

closed 2017-06-02 by hughbris

don, 2017-06-02:
Sure send a patch for the README.

Depending on what you need to do with beacons, another solutions is write some native code to do exactly what you need and make a custom plugin.