don/cordova-plugin-ble-central #412: ERROR when disconnecting

asierdanjou, 2017-06-02:
Hi, 
Sometimes when trying to disconnect I get this error:

_ERROR: Attempt to invoke virtual method 'void android.bluetooth.BluetoothGatt.close()' on a null object reference._

my disconnect function is next one:

    _disconnect: function(event) {
        ble.disconnect('00:A0:50:A3:25:9E', app.onOkDis, app.onError);
    },_

Any idea why does It catch this onError event ? 
Regards,
Asier.

acutetech, 2017-06-13:
My apologies to all for not reading this before posting #416, which is probably the same as this issue. Asier - do you see the other cases I reported? What phone(s) are you using?

asierdanjou, 2017-06-13:
I'm using a Samsung galaxy S4 with Android 5.0.1
Yes, I read the others cases. In my phone sometimes it goes as expected, and other times happens which you described at second case but peripheral always gets disconnect although it has reported error.
I do a simple code just for testing Connect and Disconnect functions.
I've tested 'cordova.exec( app.onOkDis, app.onError, 'BLE', 'disconnect', [deviceId]);' instead 'ble.disconnect(deviceId, app.onOkDis, app.onError);' but same issue happens.

don, 2017-06-14:
The disconnect implementation needs better null handling.

asierdanjou, 2017-06-14:
Any example for null handling ?

acutetech, 2017-06-21:
For me this worked: change disconnect() so it only calls gatt.disconnect(), then in onConnectionStateChange(), in the else{} case, add this:
            gatt.close();
            gatt = null;
            connectCallback = null;

ccravens, 2017-07-12:
Since I'm getting an error in iOS I'm going to open up a new issues since I don't think these are related

don, 2018-06-07:
This is one of the many disconnect problems that happens on some Android devices. Hoping #592 fixes it.

commit 37790ff6 (2017-09-05): change disconnect logic

Fixes #412
See #416 #422 #441
closed 2018-06-07 by don