don/cordova-plugin-ble-central #415: Supplied parameters do not match any signature of call target

nimeso, 2017-06-12:
I keep getting a error during the build.
'Supplied parameters do not match any signature of call `target.'`

```javascript
import { Component } from '@angular/core';
import { IonicPage, NavController, NavParams } from 'ionic-angular';
import { BLE } from '@ionic-native/ble';
@IonicPage()
@Component({
    selector: 'page-settings',
    templateUrl: 'settings.html'
})
export class Settings {
    constructor(public navCtrl: NavController, public navParams: NavParams, private ble: BLE) {
    }
    ionViewDidLoad() {
     	this.ble.scan();
    }
    onSuccess(device){
        console.log(device);
    }
    onFailure(error){
        console.log(error);
    }
}
```

Cordova CLI: 6.5.0
Ionic Framework Version: 3.0.1
Ionic CLI Version: 2.2.2
Ionic App Lib Version: 2.1.7
Ionic App Scripts Version: 1.3.0
ios-deploy version: Not installe
ios-sim version: Not installed
OS: Windows 10
Node Version: v6.10.2
Xcode version: Not installed

aaron-mile6, 2017-06-12:
That error means that you didn't supply the correct parameters compared to what the function was looking for.

If you reference the docs:
https://ionicframework.com/docs/native/ble/#scan

the scan function is looking for two parameters, services, and seconds.  At minimum you would want to do this.ble.scan( [], 5);.  But that wouldn't do anything with the results, it returns an observable that you need to subscribe to.

 `
this.ble.scan(
    [],
    5,
).subscribe(
    (device: any) => {
        console.log( device);
    }
);
`

should give you a starting point

nimeso, 2017-06-13:
I'm sooo stuck lol. I can't get anything to work and I'm pretty confused.
Here is my really simple code. What am I doing wrong? My new error is:
Property 'startScan' does not exist on type 'typeof BLE'.
Thanks for all your help.
```javascript

import { Component } from '@angular/core';
import { IonicPage, NavController, NavParams } from 'ionic-angular';
import { BLE } from '@ionic-native/ble';

@IonicPage()
@Component({
  selector: 'page-settings',
  templateUrl: 'settings.html'
})
export class Settings {

    constructor(public navCtrl: NavController, public navParams: NavParams) {

    }

    startScanning() {
		console.log('start scanning');
		BLE.startScan([]).subscribe(device => {
			console.log(device);
		});
	}

  
}
```
and my html button:
```html
<button (click)="startScanning()">Scan</button>
```

aaron-mile6, 2017-06-13:
Why did you remove ble from the constructor?

To use `@ionic-native/ble';` you need it loaded in your constructor

`private ble: BLE`

Then

`BLE.startScan([])`

Should be

`this.ble.startScan([])`

don, 2017-07-07:
@nimeso I recommend getting simple examples working with your hardware *without Ionic*. 

Once you get this working, you can port to Ionic, optionally using the BLE wrapper. Note that this plugin provides [ble.js](https://github.com/don/cordova-plugin-ble-central/blob/master/www/ble.js). The [TypeScript bindings in BLE](https://ionicframework.com/docs/native/ble/) are provided by Ionic and wrap this code.

closed 2017-07-07 by don