don/cordova-plugin-ble-central #417: IOS write while scanning fails silently

aaron-mile6, 2017-06-13:
I have an app that scans for devices, and in the scan success callback, I am connecting and writing a command.  On IOS (10.3.2) this command silently fails if the scan is still going on, simplified code snippet and log messages below ( this is Typescript running on ionic 2):

```
this.ble.scan(
	[ this.serviceUUID ],
	5,
).subscribe(
	(device: any) => {
		this.ble.connect( device.id ).subscribe(
			( ) => {
				this.ble.startNotification(
					device.id,
					this.serviceUUID,
					this.rxCharacteristic,
				).subscribe( ( data: any ) => {
					this.rawReadHandler( data, device.id );
				});
                                console.log( 'BLE - Send Getinfo');
				this.ble.write(
					device.id,
					this.serviceUUID,
					this.txCharacteristic,
					new Uint8Array( [xxx] ).buffer,
				).then(
					( data: any ) => {
						console.log('BLE - ' + deviceArrayKey + ' Send Get info: BLE Write success: ' + JSON.stringify( data));
					},
					( error: any ) => {
						console.log('BLE - ' + deviceArrayKey + ' Send Get info: BLE Write error: ' + JSON.stringify( error));
					},
				);
			},
		);
	}
);
```

> registering for notification
> getData
> Looking for 00035B03-58E6-07DD-021A-08123A000301 with properties 16
> BLE - Send Getinfo
> getData
> Looking for 00035B03-58E6-07DD-021A-08123A000301 with properties 8
> BLE - Send Get info: BLE Write success: null

The problem is that the device never receives the written command.  If I put a timeout on the write command to run after the scan time stops, then the write goes to the device successfully.

aaron-mile6, 2017-06-13:
Upon working on this further i'm not sure if it is affected by the scanning process.   Because when I put a 1 second timeout before the write, it was still going out before the scan finished, but it was successful.  So it may be due to the sequence of connecting, setting up the notification and writing immediately?

don, 2017-06-14:
You're better off writing once the scan completes.

aaron-mile6, 2017-06-14:
Thanks for the tip, and I have updated my code to reflect that, but the issue seems to be the silent failure, if the write wasn't successful, shouldn't the failure callback have been called?

don, 2017-07-07:
>if the write wasn't successful, shouldn't the failure callback have been called?
Ideally yes. In this case I don't think the native code gets an error so the plugin can't call the failure callback.

closed 2017-07-07 by don