don/cordova-plugin-ble-central #418: Scan before Connect ?

asierdanjou, 2017-06-14:
Hi everyone,
I have a question. 
I try to connect with a peripheral knowing its BDADDR but I was no able to get it. I restart phonegap and peripheral but no way, until I've made ble.scan(); before ble.connect();.
Anyone knows why ? Is it necessary to scan with central before trying to connect ?
Thank you,
Asier.

don, 2017-06-14:
You need to scan before connecting so that the plugin has the peripheral info. Just ignore the scan results if you're connecting to a known UUID or MAC address.

closed 2017-07-07 by don

lastmjs, 2017-09-28:
Can I bring this up again? Why is a scan necessary? I'm doing a scanWithOptions, and I want to connect once I've found the device I need. Why do I need to perform a scan within my scanWithOptions? I'm doing some time sensitive things and the performance hit from having to do another scan could be problematic. Thanks.

Domvel, 2018-05-30:
@don 🤔 In the most of cases I connect directly by the known device id without a scan. (id is stored in the local storage). It works "great" on android. But I also have some issues on android ble behavior. Sometimes I get an GATT Error 133 / 129 or a service is not found. Maybe I should scan before connect? But why? On connect I get the services / characteristic directory on success callback. And in the most of cases, it works. This is a bit weird.