don/cordova-plugin-ble-central #419: Connecting to Micro:Bit

fcasanellas, 2017-06-15:
Hi!

I'm trying to use the BluefruitLE example with a Micro:BIt device with  nRF51822 (https://www.nordicsemi.com/eng/Events/BBC-micro-bit)

For the moment I'm encountering with the error "Set notification failed for 6e400001-b5a3-f393-e0a9-e50e24dcca9e" after the connection and with "Failed writing data to de bluefruit ble" when I try to send a message.

I'm just starting the debug process but if sombody have any idea to help me I wouldl be very grateful. I'll publish my following steps in this issue

Thanks!

fcasanellas, 2017-06-15:
I've seen that @bluetooth-mdw, author of multiple developments for micro:bit have forked this project. Have you succed on use it with micro:bit?

bluetooth-mdw, 2017-06-15:
I did fork it but didn't do anything with the fork in the end. I have a whole series of apps, created with this plug-in for use with micro:bit. All the apps by "Bitty Software" are Cordova apps that use this plugin. See http://www.bittysoftware.com/ 

Documentation for the Bluetooth profile is here: https://lancaster-university.github.io/microbit-docs/ble/profile/ 

Note though that most services are optional and so you need to explicitly include the ones you want in your micro:bit firmware.

Your problem, I think is that you are using the UUID of the UART service and not the characteristic whose client characteristic configuration descriptor you want to update to switch on notifications. 

Try 6E400002B5A3F393E0A9E50E24DCCA9E instead.

See https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html for the detailed documentation.

don, 2017-07-07:
@fcasanellas did Martin's suggestions fix your problem?

don, 2017-08-22:
If you load the [Espurino firmware](https://www.espruino.com/Download) on your Micro:bit it will advertise the Nordic UART UUID and work with the [bluefruitle example](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle)

closed 2017-08-22 by don