don/cordova-plugin-ble-central #421: Notification bubbles up sliced in two calls to notification callback

zabojad, 2017-06-23:
I'm experiencing the following problem on Android with latest version of this plugin.

In my app, I connect to a  BLE peripheral that sends back notifications each time I write an instruction to one of its characteristics

Sometimes, that notifications gets bubbled to the cordova context in two calls to the notifications callback passed to `ble.startNotification(device_id, service_uuid, characteristic_uuid, success, failure);`.

For example, instead of `success` being called once with `[0x5A,0x10,0x00,0x8F]`, I get `success` being called twice with `[0x5A]` the first time and `[0x10,0x00,0x8F]` the second time (right after the first one).

What could be the reason for that ?

Note that I do not reproduce this issue with another application like BLEScanner for Android...

don, 2017-07-07:
@zabojad I think you're receiving 2 notifications from the peripheral.

I'd use Android's bluetooth log to see what's being sent over the wire. 

 * "Enable Bluetooth HCI snoop log" in the developer settings.
 * run your test
 * adb pull /sdcard/btsnoop_hci.log
 * use Wireshark to see if 1 or 2 notifications are being sent

zabojad, 2017-08-16:
@don yes it was indeed the peripheral's mistake. Thanks!

closed 2017-08-16 by zabojad