don/cordova-plugin-ble-central #425: Add timeout for ble.connect()

QimatLuo, 2017-07-04:
1. ble.scan(), I get a device id is AAA
2. I remove the AAA device power
3. ble.connect() the AAA device

The connectFailure will never trigger, looks it will keep trying to connect the device.
Because after long time, I power on the device, connectSuccess will invoked.
Can you support timeout option in this case?

don, 2017-07-06:
@QimatLuo are you seeing this on iOS or Android?

QimatLuo, 2017-07-07:
@don Both. I have an iPhone 6s and Nexus 5, both use same code created from Ionic. The process like:
1. Open app on iPhone, I can see BLE device on list.
2. Open app on Android phone, I can see same BLE device on list.
3. Choose the device on iPhone/Android first, and connected.
4. Use another phone to choose the same device, it will not going to success or failed function.

This is what I use: https://ionicframework.com/docs/native/ble/

don, 2017-07-07:
This sounds like something that should be fixed. I'll try duplicating and looking at solutions. It might be a bit.

timburke, 2017-07-07:
We experience this as well and are looking into mitigating it.  It looks though from the android code that its pretty straightforward to add our own timeout if connect hasn't finished in a fixed period of time, it can call `disconnect` which will terminate the ongoing connection request. 

We haven't confirmed yet whether this causes any underlying problems in the android ble stack though

jaean123, 2017-07-12:
I think this is a good idea. Sometimes the connecting takes too long before it gives error.

timburke, 2017-07-12:
In case it helps anyone else with this issue. We tried implementing a
timeout with a simple setTimeout that called ble.disconnect and it did
*not* work on Android. We tested on Android 6 and the result was that the
connection appeared to still succeed but the peripheral was no longer in
the ble plugin's list of connected devices due to the disconnect logic in
the plugin which meant there was no way to interact with or disconnect from
the peripheral without power cycling the phone's Bluetooth.

So we didn't see a safe way to add this timeout feature and abandoned it.
If anyone has a better idea though I think it would be very helpful
especially on iOS that doesn't appear to have any builtin timeout.

peitschie, 2019-06-28:
Just noting here that iOS specifically calls out that connect attempts will never timeout: https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518766-connect

On Android however it does seem to timeout for me automatically after a certain amount of time.

peitschie, 2022-07-12:
The correct procedure here (as noted in the thread above) is to call `disconnect` to terminate ongoing connect attempts. iOS and Android both avoid timeouts as there's  a common use-case where the app wants to try and connect as long as it's running (or even in the background).

The easiest way to have a connect timeout is something like the following:
```
let connected = false;
ble.connect(deviceId, (data) =>  connected = true, (err) => console.error("connection error:" , err));
setTimeout(() => {
 if (!connected) {
  // give up attempting to connect
  ble.disconnect(deviceId);
 }, CONNECTION_TIMEOUT_MS);
}
```

closed 2022-07-12 by peitschie