don/cordova-plugin-ble-central #427: Error 133 when write action

sanavilpa, 2017-07-11:
Hello,

I am getting error 133 when I try to write a long buffer in a bluetooth device. 
Previously, the code makes write actions which works correctly. So, it is something related to the data that I try to send in the specific step, maybe is it too long? is there some limit in the number of bytes of data array?
Below I attach the code related to this step:

***********************
  public apduSelect = new Uint8Array(29);
  public deviceID = "DE:08:6B:53:71:0A";
  public serviceid = "71e35e48-627b-11e3-bf13-d231feb1dc81";
  public characteristicWriteid = "fc72bf5a-de74-4134-91f2-e2120fee2bc3";
  public characteristicReadid = "027c2675-75fe-4053-9f6f-2955dd3d9ad8";

    this.apduSelect[0] =0x6F;
    this.apduSelect[1] =0x13;
    this.apduSelect[2] =0x00;
    this.apduSelect[3] =0x00;
    this.apduSelect[4] =0x00;
    this.apduSelect[5] =0x00;
    this.apduSelect[6] =0x03;
    this.apduSelect[7] =0x00;
    this.apduSelect[8] =0x00;
    this.apduSelect[9] =0x00;
    this.apduSelect[10] =0x00;
    this.apduSelect[11] =0xA4;
    this.apduSelect[12] =0x04;
    this.apduSelect[13] =0x00;
    this.apduSelect[14] =0x0E;
    this.apduSelect[15] =0x32;
    this.apduSelect[16] =0x50;
    this.apduSelect[17] =0x41;
    this.apduSelect[18] =0x59;
    this.apduSelect[19] =0x2E;
    this.apduSelect[20] =0x53;
    this.apduSelect[21] =0x59;
    this.apduSelect[22] =0x53;
    this.apduSelect[23] =0x2E;
    this.apduSelect[24] =0x44;
    this.apduSelect[25] =0x44;
    this.apduSelect[26] =0x46;
    this.apduSelect[27] =0x30;
    this.apduSelect[28] =0x31;

 
  this.ble.write(this.deviceID, this.serviceid, this.characteristicWriteid,this.apduSelect.buffer).then((deviceData)=> {
    console.log("Request:" + this.bytesToString(new Uint8Array(this.apduSelect.buffer)));
    this.ble.read(this.deviceID, this.serviceid, this.characteristicReadid).then((dataToRead)=> {
      console.log("Response:" + this.bytesToString(new Uint8Array(dataToRead)));
      var buffer = new Uint8Array(dataToRead);
      console.log("Bank Application AID: "+ this.bytesToString(buffer.slice(26,34)));

        }, error => {
          console.log(error);
        });

    }, error => {
        console.log(error);
    });

*************************

That characteristic has WRITE property. 
Please, could you help me with this issue?
Thank you.

don, 2017-07-22:
It might be you're sending to much data. Check the peripheral to see the size of the characteristic. If you have the right hardware you can define characteristics up to 500 bytes with Bluetooth 4.0 But many devices have 20 bytes max characteristic length.

sanavilpa, 2017-07-24:
Thank you Don.
I have managed to write data in block of 20 bytes(max). But, my issue now is that the response is 3 block of 20 bytes, and I can only read the last block of data. **Are there any way to read all block of data sent by the BLE device?**

Thank you.

My code now is:
  public apduSelect1 = new Uint8Array(20);
  public apduSelect2 = new Uint8Array(9);

this.ble.write(this.deviceID, this.serviceid, this.characteristicWriteid,this.apduSelect1.buffer).then((deviceData)=> {
    console.log("Request:" + this.bytesToString(new Uint8Array(this.**apduSelect1**.buffer)));
    this.ble.write(this.deviceID, this.serviceid, this.characteristicWriteid,this.apduSelect2.buffer).then((deviceData)=> {
      console.log("Request:" + this.bytesToString(new Uint8Array(this.**apduSelect2**.buffer)));
      **this.ble.read(this.deviceID, this.serviceid, this.characteristicReadid).then((dataToRead)=**> {
       
        console.log("Response:" + this.bytesToString(new Uint8Array(dataToRead)));
        var buffer = new Uint8Array(dataToRead);
        console.log("Bank Application AID: "+ this.bytesToString(buffer.slice(26,34)));

          }, error => {
            console.log(error);
          });

      }, error => {
          console.log(error);
      });

    }, error => {
        console.log(error);
    });

don, 2017-07-24:
@sanavilpa If you can startNotification on the response characteristic, that would be better than reading it. This way your callback will be called anytime data is received.

closed 2017-08-22 by don