don/cordova-plugin-ble-central #429: ble.write returns ' For input string: "0000null" '

janunezc, 2017-07-15:
```
import { Component } from '@angular/core';
import { NavController } from 'ionic-angular';
import { BLE } from '@ionic-native/ble';

@Component({
  selector: 'page-led',
  templateUrl: 'led.html'
})
export class LEDPage {
  /**
   * Constants that we will use around the controller
   */
  public constants  = {
      CMD_FIND_ANGU:"FIND ANGU",
      CMD_STOP_SCAN:"Scanning... (Click to Stop)",
      CMD_TOGGLE_LED:"CAMBIAR LED"
    };

  /**
   * Constructor for the controller. Everything begins here.
   */
  constructor(public navCtrl: NavController, private ble: BLE ) {
    this['messages'] = [];
    this['valor'] = 0;
    this['service_id'] = "19B10010-E8F2-537E-4F6C-D104768A1214";
    this['ble'] = ble;
    this['ComandoTXT'] = this.constants.CMD_FIND_ANGU;
    this['targetDevice'] = {};
    this.setMessage("Constructor: Begin!");
  }
  
  /**
   * This is triggered when the button is clicked.
   */
  public Command(){
    this.setMessage("COMMAND RECEIVED!");
    this.setMessage(this["ComandoTXT"]);
    if(this['ComandoTXT']===this.constants.CMD_FIND_ANGU) {
      this['ComandoTXT'] = this.constants.CMD_STOP_SCAN;
      this.findANGU();
    } else if (this['ComandoTXT']===this.constants.CMD_STOP_SCAN) {
      this.setMessage("Command: STOPING SCAN...");
      clearInterval(this['intervalHandle']);
      this['ComandoTXT'] = this.constants.CMD_FIND_ANGU;
    } else if(this['ComandoTXT']=== this.constants.CMD_TOGGLE_LED){
      this.setMessage("ComandoTXT = CAMBIAR LED: Calling txData()");
      this.txData();    
    }
  }
  
  /**
   * This searches for ANGU in the AIR using SCAN technique.
   */
  public findANGU(){
      this.setMessage("SCAN: Begin!");
      let  ble = this['ble'];
      
      this.setMessage("SCAN: BLE ENABLE!");
      ble.enable();
      
      this.setMessage("SCAN: Setting Interval...");
      this['intervalHandle'] = setInterval(() => { // SET AN INTERVAL THAT RUNS EVERY 3 Seconds
        this.setMessage("INTERVAL: BLE SCAN...");
        //https://ionicframework.com/docs/native/ble/ 
        ble.scan([], 2 /*seconds (0) */).subscribe( data => { //DO SCAN DURING 1 SECOND
          this.setMessage("SCAN SUBSCRIBER: " + data['id'] + ' | ' + data['name'] + ' | ' + data['rssi']);
          if(data['name']=="ANGU"){
            this.setMessage("SCAN SUBSCRIBER: ANGU FOUND! STOPPED SCANNING!");
            clearInterval(this["intervalHandle"]);
            this["targetDevice"] = data;
            this["ComandoTXT"] = this.constants.CMD_TOGGLE_LED;
          }
        });
      },2100);//END OF INTERVAL DEFINITION
    }  

  /**
   * This transmits data to the selected ANGU device  
   */  
  public txData(){
    this.setMessage("txData: BEGIN! Doing ble connect...");
    
    let id = this['targetDevice'].id;
    this.setMessage(id);
    
    this.ble.connect(id).subscribe(datos=>{
      this.setMessage("BLE CONNECT SUBSCRIBE: BEGIN. Doing ble write...");
      this['valor'] = "hello world";
      this.ble.write(this['targetDevice'].id, this['targetDevice'].service_id,this['targetDevice'].service_id, this.stringToBytes(this["valor"]) ).then(()=>{
        this.setMessage("BLE WRITE THEN!");
        this.ble.disconnect(id);
      },(error)=>{
        this.setMessage("BLE Write ERROR!");
        this.setMessage(error);
      });
    },error=>{
      this.setMessage("BLE Connect ERROR!");
      this.setMessage(error.message);
    });
  }
  
  /**
   * Adds message to the history on screen
   */
  public setMessage(message){
    var count = this['messages'].length;
    message = count + ':' + message;
    this['messages'].unshift(message);
  }

  // ASCII only
  public stringToBytes(string) {
     let array = new Uint8Array(string.length);
     for (let i = 0, l = string.length; i < l; i++) {
         array[i] = string.charCodeAt(i);
      }
      return array.buffer;
  }
  
  // ASCII only
  public bytesToString(buffer) {
      return String.fromCharCode.apply(null, new Uint8Array(buffer));
  }

}

```

janunezc, 2017-07-15:
I think I found my mistake

closed 2017-07-15 by janunezc

bitf12m553, 2020-10-21:
Hi @janunezc I know this is very old post. But can you let me know what was the issue in your case?
I am getting the same issue?

Thanks in advance!

janunezc, 2020-11-09:
Hi @bitf12m553 

That's kind of old 👍 

I don't want to make you waste your time. One tip would be to look into this repository by the dates "I found my mistake" and see what changed there.

https://github.com/janunezc/robotics/commits/master