don/cordova-plugin-ble-central #430: BLE plugin is not installed if app starts from memory

ruhl, 2017-07-21:
I have a really strange problem:

If the application starts, ble is working without problems. Then I close the application (app is still in memory) and start the app again, I get the message:
`Native: tried calling BLE.isEnabled, but the BLE plugin is not installed.
pluginWarn @ vendor.js:56455
vendor.js:56461 Install the BLE plugin: 'ionic cordova plugin add cordova-plugin-ble-central'
pluginWarn @ vendor.js:56461
Install the BLE plugin: 'ionic cordova plugin add cordova-plugin-ble-central'
pluginWarn @ vendor.js:56461
vendor.js:56455 Native: tried calling BLE.enable, but the BLE plugin is not installed.
pluginWarn @ vendor.js:56455
Native: tried calling BLE.enable, but the BLE plugin is not installed.
pluginWarn @ vendor.js:56455
vendor.js:56461 Install the BLE plugin: 'ionic cordova plugin add cordova-plugin-ble-central'
pluginWarn @ vendor.js:56461
Install the BLE plugin: 'ionic cordova plugin add cordova-plugin-ble-central'
pluginWarn @ vendor.js:56461`

If I remove the app from the list of recently used, I can start the app without problems.
On Android 4.4 this works without problems. But on Android 7.0 and also an 6.01 I have that problem.

I updated the framework and recreated the application, but nothing helped.
ionic info:
`global packages:

    @ionic/cli-plugin-proxy : 1.3.2
    @ionic/cli-utils        : 1.5.0
    Cordova CLI             : 7.0.1 
    Ionic CLI               : 3.5.0

local packages:

    @ionic/app-scripts              : 2.0.2
    @ionic/cli-plugin-cordova       : 1.4.1
    @ionic/cli-plugin-ionic-angular : 1.3.2
    Cordova Platforms               : android 6.2.3
    Ionic Framework                 : ionic-angular 3.5.3

System:

    Node       : v6.10.3
    OS         : Linux 3.10
    Xcode      : not installed
    ios-deploy : not installed
    ios-sim    : not installed
    npm        : 3.10.10 `

here is a test-repo: https://github.com/ruhl/ble_test

don, 2017-07-22:
@ruhl thanks for providing a test project. I'll look into this next week.

ruhl, 2017-07-31:
If I move the code into the constructor, it works from time to time.
But if I compile the code with --prod, same problem as before.

It seems to be a kind of race condition?

don, 2017-08-22:
@ruhl I can duplicate this with your example but adding the --prod flag.

I can also duplicate this with my [BLE Connect example]() by adding the --prod flag on iOS and Android. The off thing is the app says "Error plugin_not_installed' when it starts, but if I hit the scan button, everything works fine.

I'm guessing it's a problem with my code or some of the Angular build crap, but there could be an issue with Ionic Native or the plugin.  I'm going reach to to Ionic to see if the have suggestions.

don, 2018-06-13:
This is an Ionic issue. Hopefully fixed by now?

closed 2018-06-13 by don

Krastanov, 2018-07-16:
FYI, it does not seem to be fixed in Ionic yet.

don, 2018-07-16:
@krastanov can you create a very simple project so I can duplicate this? Provide details about the os, os version, and ionic version and steps to follow. I haven’t been able to duplicate this.

Krastanov, 2018-07-17:
Certainly!

I made a branch of a current project of mine where I deleted everything but the scanning functionality:

https://github.com/Krastanov/orb-monitor/tree/plugin_not_installed

It is one of the default Ionic templates with a single page modified to act as a scanner: https://github.com/Krastanov/orb-monitor/blob/plugin_not_installed/src/pages/connect-to-orb/connect-to-orb.ts

The scanner is started on page load and can be restarted by pressing the scan button.

Errors are caught and sent to the `console.error` and printed on the app screen as well.

(Sorry, this is hardly minimal, but I am new to Ionic and I know only how to start with one of their templates.)

What happens:

1. Run `ionic cordova run android` and everything works perfectly.
2. Run `ionic cordova run android --livereload --consolelogs` and the first load is perfect, but then after modifying the file (just adding a whitespace for instance) and waiting for the reload results in a "plugin_not_installed" message. Pressing the "scan" button works after that though, so the plugin must have actually been installed.

I would be happy to help with debugging this, but given how minor the problem is (and how it is plausible that it is in Ionic, not in this plugin), I do not want to seem like I am expecting you to spend your precious time on this. I know how much effort it is to provide a great open-source library, and I am grateful for the service you have already provided for the community.

Krastanov, 2018-07-17:
Forgot to mention: Tested on Android (7.1.2, LineageOS 14.1, OnePlus3).

Compilation done on Debian (rolling experimental).

npm --version
6.1.0

ionic version
3.20.0