don/cordova-plugin-ble-central #431: ble.write

janceymario, 2017-07-24:
Problem Description: In your plugin, I am little bit confused about the use of the ble.write method.
Ble.write (device_id, service_uuid, characteristic_uuid, data, success, failure)
 
In this method the parameter “data” is binary type, where arraybuffer should be used. In the annotations, data are sent by Uint8Array in bytes.
 
Now I have a request: our Bluetooth communication uses the hexadecimal string to do the communication. For example, the string to be communicated is: "6cab1232d4"
According to our Bluetooth communication rules, this ten-digit string is sent in hexadecimal numbers, which should be 5 bytes in size. 
 
My question: how to use the example in this method?
 
Or should I use unit8array to define 5 bytes array, and then take the buffer of them to send out?

don, 2017-07-24:
The data needs to be sent as an ArrayBuffer. 

One way to do this is to create a Uint8Array, then use the buffer property to get an ArrayBuffer.

    > a = new Uint8Array([0x6c, 0xab, 0x12, 0x32, 0xd4])
    Uint8Array [ 108, 171, 18, 50, 212 ]
    > a.buffer
    ArrayBuffer { byteLength: 5 }

janceymario, 2017-07-25:
Ok, thanks for the fast answer

closed 2017-07-25 by janceymario