don/cordova-plugin-ble-central #432: Disbound device or reconnect again to already bounded device

Samb101, 2017-07-24:
Hello,

First, thank you so much Don for this work.

It's not a bug at all but
I was wondering if there was any way to disbound a bounded device ?

Actually, I can't reconnect to a device already bounded.
Currently, I have to disbound , and then re-bound the two devices.

Maybe I pass through something.

Thank you again.

don, 2017-07-24:
I don't think there's a good way to do this. Both iOS and Android have some functions to connect to connected devices, but there's no JavaScript API in the plugin. If I recall correctly iOS has some functions to find and connect to previously bound devices.

Samb101, 2017-07-24:
Ok, thanks for the fast answer

closed 2017-08-22 by don