don/cordova-plugin-ble-central #433: Problem connecting using the UART example

rogerclarkmelbourne, 2017-07-25:
I'm trying to modify the RedBear Chat example to work with the Nordic UART example, but I get this error when I connect.

Attempt to invoke virtual method java.util.list.android.bluetooth.bluetoothgattservice.getcharacteristics on a null

I've checked the characteristic values 

https://github.com/xueliu/nRF52/blob/master/nRF52_SDK_0.9.1_3639cc9/components/ble/ble_services/ble_nus/ble_nus.c

and mine seem to be correct

```
	serviceUUID:		"6E400000-B5A3-F393-E0A9-E50E24DCCA9E",
	txCharacteristic:	"6E400002-B5A3-F393-E0A9-E50E24DCCA9E",
	rxCharacteristic: 	"6E400003-B5A3-F393-E0A9-E50E24DCCA9E"
```

Strangely the heart rate example seems to work OK, and that uses basically the same code.

Anyway, I don't think it it should be generating the error alert about the java null object, it should be calling app.onError

don, 2017-07-25:
The Bluefruit example should work with the Nordic UART service. https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle

rogerclarkmelbourne, 2017-07-25:
Hi Don

I've tried that example it doesn't work either
It works fine on Android 6, but on Android 7, I get varying problems,

The first time I ran the demo, it called app.onError, but that just alerts [object object]

The second time I ran it, I don't get that error, and I can see from the Nordic end, that it is connecting, but its not writing.

Does the plugin need to request network permission at runtime for it to be able to work on Android 7?

don, 2017-08-22:
> The first time I ran the bluefruitle demo, it called app.onError, but that just alerts [object object]

Android was listing all devices. When you scrolled, it probably connected to some devices. These alerts are devices disconnecting. The old API returned a string, the newer API returns JSON. I updated the code to stringily the JSON before showing the alert.

Versions of Android > 5.1 are a lot better at filtering when scanning. I updated the code to filter devices when scanning on Android. 

> The second time I ran it, I don't get that error, and I can see from the Nordic end, that it is connecting, but its not writing.

Not sure what's happening there. I connected to a few devices and writing was working OK. Some older Adafruit devices use property: writeWIthoutResponse instead of write, but the code is checking for that.

> Does the plugin need to request network permission at runtime for it to be able to work on Android 7?

The plugin request location permission for BLE scanning.

> I noticed in the js code that scanning for 128 bit UUID's doesnt seem to work on Android

Scanning for 128 bit or 16 bit UUIDs works on most Android 6 and Android 7 devices. But there are always problematic devices that don't work.

closed 2017-09-05 by don