don/cordova-plugin-ble-central #434: response  data

janceymario, 2017-07-25:
Question Description: Hello, I have one question when I used this plug-in. After I send data to Bluetooth, Bluetooth will reply to me with some data. My question is how can I get the response data from Bluetooth? 
1.    function: Ble.write (device_id, service_uuid, characteristic_uuid, data, success, failure). 
Should I get the Bluetooth response data from the “success” callback? 
2.    Or should I use ble.read to get the response ?

tiandongmao, 2017-08-16:
You can call ble.startNotification (device_id, service_uuid, characteristic_uuid, data, success, failure) before ble.write.

don, 2017-08-22:
With Bluetooth LE the properties of the characteristic determine if you can read, write, or notify. Like @tiandongmao says, notifications are generally the most convenient way to get data from the peripheral back to the phone. You'll need to get more info about the peripheral to determine how to communicate with it. If there's no documentation, try connecting with LightBlue Explorer to see what's available.

closed 2017-08-22 by don