don/cordova-plugin-ble-central #435: when invoke ble.scan ,failure method callback

kitikousei, 2017-07-26:
adb log

07-27 00:45:16.338 18591 18664 E Adreno-GSL: <ioctl_kgsl_device_getproperty:830>: Either IOCTL_KGSL_DEVICE_GETPROPERTY 16 unsupported by kernel or 4 bytes is the wrong size
07-27 00:45:16.340 18591 18664 E Adreno-GSL: <ioctl_kgsl_device_getproperty:830>: Either IOCTL_KGSL_DEVICE_GETPROPERTY 16 unsupported by kernel or 4 bytes is the wrong size

android version 6.0.1 MXB48T

kitikousei, 2017-07-26:
[image]

don, 2017-08-22:
What data are you passing to scan? The first parameter should be a list of UUIDs. UUIDs should be 128 bit or 16 bit Strings.

    ble.scan(['FFE0'], success, failure);

Some Android devices can't filter when scanning. If that's the case try scanning for all devices by passing an empty list.

    ble.scan([], success, failure);

closed 2017-09-05 by don