don/cordova-plugin-ble-central #437: BLE.connect return errorMessage:"Peripheral Disconnected" on Android 7.11

Dongsipan, 2017-08-05:
On android 7.11 ,always connect failed, but sometimes it works well.Android 6 all is well.  I Need Help!!

don, 2017-08-21:
That message is likely coming from https://github.com/don/cordova-plugin-ble-central/blob/3a0acb250f8bde2127bb723e23cdab0b71611ef0/src/android/Peripheral.java#L217-L235

I'm not seeing this on my Nexus 5X with 7.1.2.

Can you edit your code and log what's happening with the plugin?

Put something like

    LOG.d(TAG, "onConnectionStateChange status " + status + ", newState " + newState);

elijthomas, 2017-08-30:
I have the same issue, on the Samsung s7 , It connects sometimes but not always

don, 2017-09-05:
This message is likely correct. The device is disconnecting. It could be a problem with Bluetooth on Android or the peripheral. I'd look at the Bluetooth HCI snoop log to try and determine what's happening. https://blog.bluetooth.com/debugging-bluetooth-with-an-android-app

I'm closing this for now. If you keep seeing the error, reopen this issue (or create a new issue) with more details about the hardware you're using, the code creating the error, adb logcat output etc.

closed 2017-09-05 by don