don/cordova-plugin-ble-central #438: Multiple Characteristics with same UUID on same service

rodie, 2017-08-07:
Hey, 
I'm trying to connect an iPhone with an Bluetooth LE Mouse.  (extract):
**_{
      "properties": [
        "Read",
        "Write",
        "Notify"
      ],
      "isNotifying": false,
      "characteristic": "2A4D",
      "service": "2B080000-BDB5-F6EB-24AE-9D6AB282AB63"
    },
    {
      "properties": [
        "Read",
        "Write",
        "Notify"
      ],
      "isNotifying": false,
      "characteristic": "2A4D",
      "service": "2B080000-BDB5-F6EB-24AE-9D6AB282AB63"
    },
    {
      "properties": [
        "Read",
        "Write",
        "Notify"
      ],
      "isNotifying": false,
      "characteristic": "2A4D",
      "service": "2B080000-BDB5-F6EB-24AE-9D6AB282AB63"
    },
    {
      "properties": [
        "Read",
        "Write"
      ],
      "isNotifying": false,
      "characteristic": "2A4D",
      "service": "2B080000-BDB5-F6EB-24AE-9D6AB282AB63"
    },_**
    

The service (2B080000-BDB5-F6EB-24AE-9D6AB282AB63 ) has multiple characteristics with the same UUID.  Using the startNotification Method with service UUID and the characteristic UUID only returns notifications for the first entry, I'm not able to get any events for the other entries.
To verify the issue I used the LightBlue app on iOS. It lists all characteristics and I'm able to subscribe them get individual events of each characteristic.

Is there any way to handle multiple characteristics with the same uuid and service on ble central?
Thanks for your help!

don, 2017-08-22:
@rodie That's very odd that the mouse is re-using the same characteristic with the same properties in a service. Generally you'd want to treat a Bluetooth mouse a Human Interface Device (HID) rather than connect directly to it, but I guess the iPhone doesn't support the mouse.

When you start the notification, the plugin creates a key based on the device id, service id, and characteristic id and stores the callback using this key. https://github.com/don/cordova-plugin-ble-central/blob/2a3d8b95b094647f22a90b92609d77d93b1bb4b2/src/ios/BLECentralPlugin.m#L186-L188

The actual notification is enabled by `peripheral setNotifyValue:forCharacteristic:` https://github.com/don/cordova-plugin-ble-central/blob/2a3d8b95b094647f22a90b92609d77d93b1bb4b2/src/ios/BLECentralPlugin.m#L191

If you modify `startNotification` to call `setNotifyValue:YES` multiple times it *might* work. I'd try iterating through the characteristics and call `setNotifyValue` if the characteristic UUID matches your UUID.

If you're OK with all notifications going to the same callback, the existing hooks back to Cordova should work. If you need separate callbacks for each notification, you'll need to come up with a new scheme for creating keys. Unfortunately [CBCharacteristic](https://developer.apple.com/documentation/corebluetooth/cbcharacteristic?language=objc) doesn't expose the attribute handle.  https://github.com/don/cordova-plugin-ble-central/blob/2a3d8b95b094647f22a90b92609d77d93b1bb4b2/src/ios/BLECentralPlugin.m#L805-L807

closed 2017-09-05 by don

drewcovi, 2018-05-01:
@don Similar issue with identical characteristic UUIDs in one service. Does this plugin use handles so those characteristics can be easily found? Seems this is how webbluetooth handles connection.

don, 2018-05-01:
@drewcovi I don't think there's any way to get the handles from iOS or Android APIs.

https://developer.apple.com/documentation/corebluetooth/cbcharacteristic
https://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic

truongsinh, 2019-03-11:
@don 
Actually, for Android, there's https://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic.html#getInstanceId() 

I'm still looking on iOS side though...