don/cordova-plugin-ble-central #440: Calling the writeWithoutResponse method on Android 7.0 encountered an error

tiandongmao, 2017-08-16:
Hello,
I encountered an error when calling the writeWithoutResponse method on Android 7.0,
Steps in the program:
1. Connect to a peripheral
2. Call the writeWithoutResponse method to send a message
3. Receive data from the peripheral
4. Immediately call the writeWithoutResponse method again to send the same message, I encountered an error.The error message is "Write failed".

But all is well on android 6. 
Thank you for any help.

don, 2017-08-22:
Can you check the properties on your characteristic and ensure that has writeWithoutReponse and not write?

tiandongmao, 2017-08-22:
Yes,here is the screenshot of my code.

[image]
[image]

tiandongmao, 2017-08-29:
Today, I found that if I have called the **stopNotification** method,
the writeWithoutResponse method will never success on android 7.x.

Finally ,I resolved the problem by removing the calling of **stopNotification** method.

Thanks @don !

closed 2017-09-05 by don

woodz-, 2018-08-06:
I am not happy with the workaround taken. It's fair enough if you can live without stopping notifications. But I am facing a case where I need to stop notifications before I can do a write request, then capturing all incoming data via active notifications. It turns out that write requests are not allowed while actively listen to notifications (via the startNotification(...) method).

If I'm running the application without ever starting notifications, write requests are working fine. If I am running with notifications active, then stop them to gain permission for write requests, any such write request fails with the same message as @tiandongmao encountered: "Write failed". There is not much clue to get out of this message btw.

Due to the properties of the peripheral in use:
`{
  "name":"BT12",
  "id":"00:15:83:E0:2E:C6",
  "advertising":{},
  "rssi":-73,
  "services":["ffe0","1800","180a"],
  "characteristics":[{
      "service":"ffe0",
      "characteristic":"ffe1",
      "properties":["Notify"],
      "descriptors":[{"uuid":"2902"}]
    },{
      "service":"ffe0",
      "characteristic":"ffe2",
      "properties":["WriteWithoutResponse","Write"]
    },{
      "service":"ffe0",
      "characteristic":"ffe3",
      "properties":["Read","Write"]
    },{
      "service":"1800",
      "characteristic":"2a00",
      "properties":["Read"]
    },{
      "service":"180a",
      "characteristic":"2a24",
      "properties":["Read"]
    },{"service":"180a",
      "characteristic":"2a29",
      "properties":["Read"]
    },{
      "service":"180a",
      "characteristic":"2a26",
      "properties":["Read"]
    },{"service":"180a",
      "characteristic":"2a23",
      "properties":["Read"]
    }
  ]
}`

the service id I am using is: `ffe0`
the characteristic id for notifications is: `ffe1`
the characteristic id for write requests is: `ffe2`

While working on Android 7.1.2 as a next step I will go for Android 5.0 as a test (and IOS of course where things work a lot smoother) then see how things behave. Will keep you informed.

Greetings woodz

don, 2018-08-06:
@woodz- you should be able to write when listening for notification, unless there's something in the peripheral that doesn't allow this. For most devices you should be able to stop and restart notifications. You should be able to write after stopping notifications. I'm guessing this is a peripheral or specific android device problem, but it could be an bug. If you think it's a bug, create a very simple app, instructions to duplicate it, and open a new issue.

woodz-, 2018-08-06:
Thanks @don for your feedback. Yes, I can agree, doing write requests is indeed possible during active notifications. The problems denoted above occur if I start switching repeatedly between notifications off - write - notifications on. The latter isn't necessary as I have learned tonight.

The trick was to chain together the callbacks of write and reception via their notifications and let them trigger each other in a reentrant way. If none of them get called within 300ms, it's a signal that the command / response conversation has finished. This timeout can be re-triggered e.g. on any single reception.

It really works smoothly and reliable now like earlier with your classic bt plugin. The world of ble is much harder to overcome. And my experience is, you need to be prepared for surprises on Android.

Many thanks, greetings woodz