don/cordova-plugin-ble-central #441: disconnect ble cause null exception

wzhsh90, 2017-08-16:
08-16 10:41:08.539 25772-25918/com.cheweifei.app E/PluginManager: Uncaught exception from plugin
                                                                  java.lang.NullPointerException
                                                                      at com.megster.cordova.ble.central.Peripheral.disconnect(Peripheral.java:89)
                                                                      at com.megster.cordova.ble.central.BLECentralPlugin.disconnect(BLECentralPlugin.java:353)
                                                                      at com.megster.cordova.ble.central.BLECentralPlugin.execute(BLECentralPlugin.java:166)
                                                                      at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:117)
                                                                      at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:98)
                                                                      at org.apache.cordova.PluginManager.exec(PluginManager.java:132)
                                                                      at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:57)
                                                                      at org.apache.cordova.engine.SystemExposedJsApi.exec(SystemExposedJsApi.java:41)
                                                                      at com.android.org.chromium.base.SystemMessageHandler.nativeDoRunLoopOnce(Native Method)
                                                                      at com.android.org.chromium.base.SystemMessageHandler.handleMessage(SystemMessageHandler.java:27)
                                                                      at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                      at android.os.Looper.loop(Looper.java:136)
                                                                      at android.os.HandlerThread.run(HandlerThread.java:61)

this caused by disconnet device  concurrently.

don, 2017-08-22:
I'm trying to duplicate this. What do you mean by "caused by disconnect device concurrently."?

don, 2017-09-05:
Duplicate of #412

closed 2017-09-05 by don