don/cordova-plugin-ble-central #443: Bonding fails on connect (Android) - Error Codes: #133, #137

d1mroberge, 2017-08-17:
When we attempt to connect to a device, and then read its data, we are able to connect but then get kicked out. The status errors we get are #133 and #137. However, when we proceed to pair the device in the Bluetooth settings first, and then attempt to connect and read device data, everything works. 

We believe the issue has to do with the fact that no bond is created when you connect. Is this a fair assumption? Has this issue been seen by anyone else?  

After looking at the android src code for the plugin, it seems like there is no implementation of the createBond() function.

don, 2017-08-22:
@d1mroberge That sounds right. Most BLE devices use "Just Works" so don't require pairing. Some devices require elevated permissions for encrypted read or write and will initiate a bonding process. IIRC I've had this work with the plugin. Maybe it's devices specific? Unfortunately I don't have many test devices that require bonding.

d1mroberge, 2017-08-22:
@don thanks for getting back to us. Have you considered adding createBond() as a feature, or trying to pull it in from the evothings implementation that is already supported?

don, 2017-08-22:
Yes, it's be nice to have createBond(), getBondedDevices() in the plugin. I haven't needed it for a client yet so it's not there.

d1mroberge, 2017-08-24:
Ok thanks. One more question, it seems that the connection is iffy when the app is run in the background or when you leave the connection for longer than a few minutes. Have you noted this behavior? For our project, sometimes a connection may need to be held for 8 hours. Is this type of behavior achievable with the plugin?

don, 2018-06-13:
ble.bondedDevices() is implemented. Created #605 for ble.createBond().

closed 2018-06-13 by don