don/cordova-plugin-ble-central #444: I can not read characteristic from thermometer

evilword666, 2017-08-18:
I am reading the characteristics of a thermometer, everything is fine, I can read the manufacturer, model and device name, 

[image]

However, when I try to read the temperature characteristic this returns a value with a coding that I do not know. For example, for when the temperature that marks the thermometer is 25.1 what brings me back is this:

[image]

Could you help me solve this problem?

This is my code for read characterictic:

`
        $scope.attend = function(){
          alert("***** SENT VALUES *****: \nMAC ADDRESS:"+ $stateParams.id+"\nID_SERVICE: "+service_id+"\nID_CHARACTERISTIC: "+characteristic_id);
          ble.read(
            $stateParams.id,
            service_id,
            characteristic_id,
            function(response){
            //lert("Value Response: "+response);
    ////////////////////////////////////////////////////////////////////////////////////////////////////
            var dbufferArray = String.fromCharCode.apply(null,new Uint8Array(response));
            alert("Response: "+dbufferArray);
            console.log("Response: "+dbufferArray);
    ////////////////////////////////////////////////////////////////////////////////////////////////////

`

I based this example:

[Ionic bluetoothLE app][3]

  [1]: https://i.stack.imgur.com/ShbdB.png
  [2]: https://i.stack.imgur.com/E0cP3.png
  [3]: https://www.sitepoint.com/communicating-with-bluetooth-low-energy-devices-in-cordova/

don, 2017-08-18:
You need to get more information about the format of the data from the Bluetooth service. Hopefully this is documented. If the thermometer is sending a floating point number try `new Float32Array(response)`.

evilword666, 2017-08-21:
Hi ,@don I tried your solution, but I obtain the next error:

[image]

don, 2017-08-21:
Since the number of bytes is not a multiple of 4, it's not a Float32Array.

Ideally you'll find some documentation from the manufacturer telling you how to decipher the data. If not print out the raw bytes as hex and see if you can make sense of the data. It might be text, it might be multiple values concatenated together.

evilword666, 2017-08-21:
This is the API I was sent with my thermometer.
[https://drive.google.com/open?id=0B1OLQp2V8ILsRTljeVk0ZjR2bGM](url)

To know the characteristics and services of my thermometer use an application called "BLE Scanner". on PlayStore.

I have converted to hexadecimal but I do not see relation with the numeric value of the temperature. For example for a temperature of 35.1 the hexadecimal value that returns me is as follows:
0x06,0x4e, 0x0c, 0x00,0xfe, 0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x04

[image]

don, 2017-08-21:
The docs say you can read using the "Health Thermometer Service". If that's true, use the Bluetooth docs to write a de-coder. Since this is a standard, someone may have already done this.

https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.health_thermometer.xml
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.temperature_measurement.xml

It also talks about reading via the custom service writing to 0xFFB1 and waiting for a notification. Section 10 looks like it has the logic to decode.

> 10、0x79
> Function : Read the Offline storage temperature data
> Send : 0x79 00 00 00 00 00 00 00 00 00 00 00 00 00 00 CRC Parameter: N/A
> Return : 0x79 AA BB CC DD EE FF GG HH II JJ KK LL MM 00CRC
> AA=0x01 : The command completed successfully
> AA=0x00 : Command execution failure or CRC check errors
> BB CC is Temperature record index ,two bytes , Low byte is in front
> DD EE FF GG is time , Low byte is in front , unit is second , It is a relative time , Benchmark for at 0:00:00 on January 1, 2000 .
> HH II JJ KK as Temperature expand 10 times , Low byte is in front , for example : 37.5°C as 370
> LL MM as Temperature record the total number , Low byte is in front

evilword666, 2017-08-21:
I did the test with the demo application of bluetooth low energy of the android developers page and I see that I can recover the temperature with a format UINT16.

[image]

So I do the conversion with that format, but I get this error:

[image]

don, 2017-08-21:
Since you have 13 bytes, try creating a  [DataView](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/DataView) from the ArrayBuffer. Then accessing the temperature by index with [getInt16](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/DataView/getInt16).

closed 2017-09-05 by don

evilword666, 2017-09-05:
Thanks, I was able to get the data with the hexadecimal format of the second and third octet