don/cordova-plugin-ble-central #445: BLE startNotification callback is not working in android device

debugger200, 2017-08-19:
Using the latest version of this plugin with ionic v3.7 central plugin is completely working fine in iOS device, unfortunately not able to register notifications for characteristics value changes. 
After I connected to my peripheral, i register for custom characteristic notification for receiving voltage value but I didn't receive callback either success or failure.. as I gone through few issues regarding plugin not working in android 6 or later but it seems to be fixed now. Even I turned on device location, no hope. Please suggest any other workaround to go forward.

LuanNg, 2017-08-20:
yes, me too. it's not working on android.

debugger200, 2017-08-20:
However ble.startNotification() method had to be subscribed before we write data. This subscription had to provide either success or failure call back. But I'm able to connect to device using connect(). I subscribe to characteristic changes using the above method once the deceive connected. Are you able to get callback on this startNotification() method ?

debugger200, 2017-08-21:
Hello Don! After debugging with my code, i find failure callback is working for other services and characteristics which i do not expect for Notifications. So this means notification() method is properly working but the peripheral characteristics (with properties: notify) does not working in android. Any other idea how to make this work on android? Since iOS device works fine.

don, 2017-08-22:
This might be a problem in the [Ionic Native wrapper](https://ionicframework.com/docs/native/ble/)?

Notifications are working OK on my Nexus 5X with a very basic Cordova app that connects to a TI Sensor Tag and shows a toast when the button state changes.

Nexus 5X with Android 7.1.2
Cordova 7.0.1
Node v6.11.2
macOS 10.12.6

don, 2017-08-22:
Verified notifications work with Ionic 3.9.2 and 3.7.

I have an [Ionic Button example](https://github.com/don/ionic-ble-examples/tree/master/button) that receives notifications from a TI Sensor tag. There is code to build a fake sensor tag with a Arduino 101. https://github.com/don/ionic-ble-examples/tree/master/arduino or you can edit the Ionic app to work with your code.

don, 2017-08-22:
I ran some more tests. The error callback is called on both Android and iOS when I call startNotification on a characteristic that does *not* have the notify property.

What is the error message when the Promise is rejected?

debugger200, 2017-08-22:
@don  Thank you for your comments! Yes. Error call back is fine in both platforms. The problem with notify property not working in android device. We dont get any error message since the promise is not rejected.

debugger200, 2017-08-22:
@don Thank you so much. It is actually our fault on writing wrong data, hence plugin does not notify us. Again thanks for such nice plugin and appreciate your work. @LuanNg Thank you! it is correct we dont get notified if we write wrong data.

closed 2017-08-22 by debugger200

don, 2017-08-22:
Glad you found the problem. Thanks for the update.

anjildhamala, 2017-10-17:
@debugger200 how are you writing wrong data and it works on ios but not on android? I have run into similar problem and it'd be helpful if you could share your problem and solution.

debugger200, 2017-10-18:
@anjildhamala I found this problem after review the code that conditional statement falling to else spot where we write different data for other ble device. The same run properly in iOS as it is fall down to success case of if. So the conclusion that  problem purely resides in our code logic, not with Ble code.

subratpalhar92, 2020-09-07:
make sure you are using write() instead of writeWithoutResponse(), this solved my issue on ios