don/cordova-plugin-ble-central #453: errorMessage "Peripheral Disconnected" on Android

luposlip, 2017-08-25:
Hi Don!

I use your great plugin for an app that's published for both Android and iOS.

Some time ago my Android app began to disconnect after 30-35 seconds after reconnecting. I thought it was because of a firmware issue, so I coded my way around it, so that the users didn't get too interrupted by it (auto reconnecting etc.).

I just realized that this is NOT an issue on iOS, so my best guess right now is that the error is caused by something in the BLE plugin.

I use your newest release plugin `1.1.4`, and Cordova version `7.0.1` (for both Android and iOS of course).

Have you experienced this before, or do you have any idea why this happens?

Any advice is highly appreciated!

Thanks.

Best,
Henrik

luposlip, 2017-08-25:
UPDATE: I even get the 30-35 seconds disconnect with nrfConnect - so it's definitely NOT related to you plugin. Closing!

closed 2017-08-25 by luposlip

don, 2017-08-25:
Unfortunately it sounds like an Android issue.