don/cordova-plugin-ble-central #457: Android Disconnect Issue

ccravens, 2017-09-11:
We are facing an issue where calling disconnect() does not physically disconnect the ble connection. This causes our app to think the device is disconnected, but in reality the device remains connected and does not broadcast. The only way to resolve this is to kill our app to release the connection and restart the app so that we see the broadcast and can re-connect.

In the plugin android source, the disconnect method is mapped to the following method:

```java
    private void disconnect(CallbackContext callbackContext, String macAddress) {

        Peripheral peripheral = peripherals.get(macAddress);
        if (peripheral != null) {
            peripheral.disconnect();
        }
        callbackContext.success();

    }
```

I've noticed that in the case the peripherals HashMap no longer contains a handle to the peripheral (peripheral is null), then the disconnect will always return success even though no disconnect() has been called on the peripheral.

Is anyone else experiencing this issue? Is it possible to instead ask the Android SDK if the peripheral is connected rather than relying on this (potentially unreliable) data structure to maintain a perfect state of peripherals that are connected?

I noticed that restarting a scan causes this peripherals data structure to be cleared of any device that is either not connected or not connecting:

```java
        // clear non-connected cached peripherals
        for(Iterator<Map.Entry<String, Peripheral>> iterator = peripherals.entrySet().iterator(); iterator.hasNext(); ) {
            Map.Entry<String, Peripheral> entry = iterator.next();
            Peripheral device = entry.getValue();
            boolean connecting = device.isConnecting();
            if (connecting){
                LOG.d(TAG, "Not removing connecting device: " + device.getDevice().getAddress());
            }
            if(!entry.getValue().isConnected() && !connecting) {
                iterator.remove();
            }
        }
```

ccravens, 2017-09-19:
I have confirmed that this issue was due to my app creating multiple connections to the same MAC address. On disconnect only one of those connections was released, leaving the other connection remaining.

closed 2017-09-19 by ccravens

don, 2017-09-19:
Thanks for the update