don/cordova-plugin-ble-central #458: IOS: isEnabled, isConnected, connect,... Not firing on call

kaddyadriano, 2017-09-13:
Thank you for the plugin! Works well on android however I am having problems with IOS. isEnabled, isConnected, connect and maybe more functions are only being triggered when I slide up the 'quick options menu'. 
[image]

I am using iOS 10.3.3. Please let me know if there is a solution for the methods to fire when they are called without waiting on sliding up the menu! Thank you in advance!

snow-dev, 2017-09-13:
Same problem here, and I should build fro production on Friday :(!

kaddyadriano, 2017-09-20:
@don Anything about this issue?

don, 2017-09-21:
@kaddyadriano I haven't had time to look into it. It would help if you could create the simplest possible app that duplicates the issue and share it via a GitHub.

snow-dev, 2017-09-21:
Ok, tomorrow I can work on that example :).

kaddyadriano, 2017-09-28:
@don I have created a test app that you can look at [BLE Test App](https://github.com/kaddyadriano/cordova-plugin-ble-central-test). Thanks in advance!

kaddyadriano, 2017-10-06:
@don anything?

kaddyadriano, 2017-10-24:
@don I have looked at the code and there is a deprecated constant https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerstate it's being used inside cordova-plugin-ble-central/src/ios/BLECentralPlugin.m on line 51 - 56
It's also used in online 221 and 223 inside isEnabled method. There may be other places.
It was replaced by https://developer.apple.com/documentation/corebluetooth/cbmanagerstate
I hope that helps.

kaddyadriano, 2017-12-15:
I have figure out what the issue was. Not the plugin but the CSP was missing ```gap:```. Here is what my CSP looks like:
```html
<meta http-equiv="Content-Security-Policy" content="default-src * 'unsafe-inline' file://* *; script-src 'self' 'unsafe-inline' 'unsafe-eval' *; font-src * data:; style-src 'self' 'unsafe-inline' *; media-src *;img-src 'self' data:; child-src * ; frame-src * gap: 'unsafe-inline'" />
```

don, 2018-01-16:
@kaddyadriano thanks for the update. Nice work tracking down the missing `gap:` in the CSP.

closed 2018-01-16 by don