don/cordova-plugin-ble-central #459: Write fails after many hours connected

Rlopez, 2017-09-16:
hi there!
I am coming across this issue where this call:

```
var ecb = setTimeout(errorCallback, 5000);
ble.writeWithoutResponse(this.connectedDevice.id, this.connectedDevice.service, this.connectedDevice.cSPP, stringToBytes(string + '\r'),
	function(data){
		clearTimeout(ecb);
	}.bind(this),
	function(error) {
		console.error("couldn't write due to " + error);
		clearTimeout(ecb);
		errorCallback();
	});
```
eventually fails after many hours (about 80 or so). As you can image it is a hard thing to repro.
Once I get that error there's only two ways of getting back to a working connection: either restarting the activity or turning the BT on and off on the Android device.

My usage looks like this:
 -connect to remote device (a HM-11 module)
-sign for notifications
-do a write, which the HM-11 replies to (every 6 seconds or so)

As expected, writes don't go throw when I get this error, which triggers a sequence where I stop notifications and call disconnect. Maybe there's something else I can do to properly clear the state of the plugin?

thanks!

don, 2017-09-19:
@Rlopez this is going to be tough to debug. I'm seeing other cases where the peripheral disconnects and it's difficult to create a new connection. If I find a solution to that, it might help you reconnect.

If you enable the [Bluetooth HCI snoop log](https://blog.bluetooth.com/debugging-bluetooth-with-an-android-app) on your phone, you can use Wireshark to help to determine if the problem is the app disconnecting or the peripheral disconnecting.

A possible ugly work around might be to have the phone disconnect and reconnect every X hours to avoid this.

Rlopez, 2017-09-22:
Thing is, I do get to re-connect, I just don't seem to be able to write, so I get no replies and my protocol times out and tries to reconnect later (which succeeds as well, but no writes, etc, etc)
I have tried to disconnect/reconnect as well (which would be a great solution for my scenario) but I came across another issue where, once every so often a disconnect would hit a null in the Java code, seemingly related to https://github.com/don/cordova-plugin-ble-central/issues/412
I was connecting/disconnecting at 20 second internals, maybe that's too short for some reason?

ddugue, 2017-10-20:
If it was on android, this PR might fix it: #481

It would be due to the queue of commands in android not getting emptied on a bad write (even after a disconnection)

Rlopez, 2017-11-08:
hi @ddugue!
Tried the changes (copied the java files over my local ones and did a Cordoba build, which AFAIK call Graddle and so on), got the same issue (this time) after four hours :)
Doesn't seem like it is significantly longer than what it used to fail before.
And, as expected, once hitting the "can't write" error no amount of reloading the add did the trick, closing the app and opening did, though.

ddugue, 2017-11-08:
Hello!

Hummm, with my code, try to call disconnect BEFORE retrying to connect. It would clear the improperly cleaned gatt and the command queue.

If that doesn't work, I guess your next step would be to try to catch an error with adb logcat. If you find one, feel free to post the trace here so maybe we can help you!

Good luck!

Rlopez, 2017-11-10:
@ddugue, @don Wow!
I still get the occasional "can't write" error, but doing the closing and then reconnecting does the job. It's been running for about 28 hours doing a ping every ten seconds, still ticking!
Thanks and thanks again!
Trunk next? ;)

Rlopez, 2017-11-16:
Re-opening this as it probably only makes sense to close this issue once the solution (if indeed @ddugue 's solution is the right thing) is in the trunk

closed 2018-06-07 by don