don/cordova-plugin-ble-central #460: BLE Enable / Disable Notifications

ccravens, 2017-09-19:
Would like to have a notification if Bluetooth is disabled or enabled while the app is in use.

**Example:**
While user is using the app, the user disables bluetooth on the phone. The app receives a notification from the cordova-plugin-ble-central plugin that bluetooth has been disabled.

While user is using the app, the user enables bluetooth on the phone. The app receives a notification from the cordova-plugin-ble-central plugin that bluetooth has been enabled.

don, 2017-09-19:
@ccravens try [ble.startStateNotifications](https://github.com/don/cordova-plugin-ble-central#startstatenotifications). It should do what you need.

ccravens, 2017-09-21:
@don Thank you for pointing that out! I don't know how I missed that in the docs

closed 2017-09-21 by ccravens